pa-pa/AskSinPP #2: Pairing issue

Xento, 2017-01-17:
When I try pairing the HM-ES-TX-WM sketch I have to trigger pairing many many times until it works.
Sometimes I can the some dataexchange but it's not pairing.
Sometimes I got the message from Homematic that I want to pair a device over the serialnumber and I have to enter the securitycode.

When I then change the type of the sensor I have to trigger the update many many times, too (config button)

pa-pa, 2017-01-18:
Which hardware/software are you using?
I using HM-MOD-RPI-PCB and HM-CFG-USB with FHEM. This works very nice.
Do you have tried also other examples? I would be good to know if there is a problem with the specific device or with the base code of the library.
Can you please record the debug output of the serial console when you try to pair or change registers.

Xento, 2017-01-21:
I found the bug ...
I connected GDO2 to Pin 3 on the Arduino, I changed it to GD0 and now it works.
By the way, does the sketch work with an pro mini 8mhz?
I'm only getting weired characters as output.

pa-pa, 2017-01-21:
Yes - I always use 3.3V Pro Minis. May be the baud rate is wrong ?

Xento, 2017-01-22:
Works now, don't know why I got garbage

closed 2017-01-22 by Xento