pa-pa/AskSinPP #13: Which TSL2561 library

Xento, 2017-08-12:
Which TSL2561 library are you using for the PIR example?

pa-pa, 2017-08-13:
Good Question. I copied it from some other project some time ago. I think it is this

https://github.com/sparkfun/SparkFun_TSL2561_Arduino_Library

Xento, 2017-08-17:
I took this library and changed the code to work with it.
At the moment it is untested but compiles:
https://github.com/pa-pa/AskSinPP/pull/17

pa-pa, 2018-02-25:
Master uses this now https://github.com/adafruit/TSL2561-Arduino-Library

closed 2018-02-25 by pa-pa

ralf33609, 2018-03-23:
Maybe this could be mentioned in readme file, because there are two libraries in adafruit github repository. The arduino library manager 1.6.5 and 1.8.6 only offers the tsl2561 unified. So you have to include the right library by importing the zip file from repository!

jp112sdl, 2018-03-23:
It is already mentioned:
https://github.com/pa-pa/AskSinPP/blob/9ec09585667457832dfbfbc33c2846e9d4a483a8/sensors/Tsl2561.h#L11

pa-pa, 2018-03-23:
And also in the [Readme.md](https://github.com/pa-pa/AskSinPP#optional-required-sensor-libraries) including links to the git repositories