pa-pa/AskSinPP #19: HM-LC-SWX-SM wrong channel count

Xento, 2017-09-17:
When you define that it should be a HM-LC-SW2-SM or a HM-LC-SW1-SM it still sends 4 channels to the CCU when pairing.
I think this is because the channels are registered at object creation and not deregistered when the channel count is reduced.
Maybe it could be resolved with conditional compiling and different object creators.

Xento, 2017-09-18:
I got it. 
Its the device Info which has tobbe changed. 
Maybe more comments which defines have to be changed would be nice

pa-pa, 2017-09-18:
ChannelDevice overloads getDeviceInfo and overwrites the real channel count inside the buffer. But this is not working because the getDeviceInfo method is not virtual - so it is never called by the Device class. We need to make it virtual. Can you please try it ?

pa-pa, 2017-09-18:
I moved the channel count into the base Device class. So the getDeviceInfo will always updated with the correct channel count. In Master I will remove the channel count from the DeviceInfo structure.

closed 2017-09-18 by pa-pa