pa-pa/AskSinPP #24: HM-SEC-RHS: no cycleInfoMsg

jp112sdl, 2017-12-30:
Hi! What a great project!

But I've got a (hopeful small) issue:

I'm using the [HM-SEC-RHS.ino](https://github.com/pa-pa/AskSinPP/blob/master/examples/HM-SEC-RHS/HM-SEC-RHS.ino) and after ~1 day the CCU says "Communication Error" with that device.

I took a look at the code and found that in line [L100](https://github.com/pa-pa/AskSinPP/blob/master/examples/HM-SEC-RHS/HM-SEC-RHS.ino#L100) the call of `cycleInfoMsg(false);` is commented out.

But whether with leaving it as it is or setting `cycleInfoMsg(true);`, the "Communication Error" appears.

pa-pa, 2017-12-30:
You need to set it to true by the CCU expert menu.

jp112sdl, 2017-12-30:
In the CCU, the option is enabled. (It is enabled by default afaik... because I did not enable it).

[image]

I will try it with a reset of the device 👍 
Or is it enough to disable and re-enable the cyclic message in the CCU (because the configuration data is retransmitted to the device)?

pa-pa, 2017-12-31:
Disable and enable again should work too.

jp112sdl, 2017-12-31:
Ok. What I did so far:
- removed the device from the ccu
- pressed D8-Button until RESET was written in the serial terminal.
- added the device to the ccu again (cyclic message was enabled by default)
- disabled cyclic message (and transfered the configuration data (short press D8))
- enabled cyclic message (and transfered the configuration (short press D8))

[image]

Now I have to wait until next year ;)

jp112sdl, 2018-01-01:
"Communication error" appeared again, just now...

jp112sdl, 2018-01-01:
At 12:13 the "Communication error" appeared and disappeared automatically at 13:25 o'clock.

I can't image where it comes from.
The rf_rhs.xml on the CCU has a cyclic message timeout of 88200 seconds, the Threestate.h should send a cyclic message every 86400 seconds.

Maybe the timer of the ATM328 has such a great abnormal variation in its timing? 

I have flashed a second, new Arduino Pro mini, with the same code (but other Serial and Device ID). There is exactly the same issue.

jp112sdl, 2018-01-01:
I've found this [link](https://forum.mysensors.org/topic/1941/gw-sleep-isn-t-very-accurate-on-a-pro-mini-3-3v-8mhz) via Google. 
Maybe the internal timing is indeed not accurate on the Pro Mini (8MHz)!

pa-pa, 2018-01-01:
You may try to short the time by change the CYCLETIME define to a lower value.

jp112sdl, 2018-01-04:
For testing, I had to set the cyclic message interval to 16 hours in Threestate.h:
`#define CYCLETIME seconds2ticks(60UL * 60 * 16)` _(That's about 57600 seconds.)_

I figured out, that the Arduini Pro Mini (3.3V / 8MHz) now sends the cyclic message after 65160 seconds (~18 hours).

That's a crazy offset of nearly 20% (2h6min. in 16 hours).

If no one else has this problem, there is nothing to do in your repository. I will change the value in Threestate.h for me.

Thanks and keep it up!

closed 2018-01-04 by jp112sdl

pa-pa, 2018-01-04:
We using the watchdog for deep sleep. It is not really exact. You can replace the 8Mhz crystal by an 32.678kHz crystal and then use the RTC class for cyclic message sending. But this also costs some extra energy.