pa-pa/AskSinPP #25: HM-ES-TX-WM inputs

baennjer, 2018-01-06:
Why does the HM-ES-TX-WM have 2 inputs? 
Only pulses from input A1 seem to be counted in the CCU while pulses on input A0 are not being counted. 
Allthough the counter values of both inputs are being transmitted to the CCU (monitored via IOBroker) only the value that is transmitted in second position is recogniced in the CCU.

pa-pa, 2018-01-07:
This is not a complete implementation of the device. 
I'm not complete sure, but this depends on the type of counter. IEC type counter will come on channel 2 all other types are in channel 1. But I only tested with FHEM - no idea how it works with CCU.

baennjer, 2018-01-07:
Will there be efforts to implement the device completely?

pa-pa, 2018-01-07:
No - this was only a proof of concept. I do not need it.

jp112sdl, 2018-09-28:
@baennjer : The CCU2-compatible version can be found [here](https://github.com/jp112sdl/Beispiel_AskSinPP/tree/master/examples/HM-ES-TX-WM_CCU) (simply switched to a lower firmware version and removed the second input channel).

closed 2018-10-26 by pa-pa