pa-pa/AskSinPP #29: Is everything available to builde a thermostat?

knopserl, 2018-02-25:
I have hot now many STM32F103 boards and what to build a thermostat such as the Homematic HM-TC-IT-WM-W-EU which has temperatur and humidity and can send back control commands. I have copied at the the channels and parameters from the homatic user guide which describes all the parameter.
I have used the stm32 example HM-LC-SWX-SM and at least the pairing with Homegear works.
I'm wondering if I have everything as bases clases which are required tuild a Thermostat like the HM-TC-IT-WM-W-EU? And are there somewhere the device models (not types) defined or can I use any numbers (in the weather sensor sample {0x00,0x3d},  is used, in the switch model has a name HM_LC_SW4_SM is used).

This are the channel "definitions" from homematic for the thermostat HM-TC-IT-WM-W-EU. Has anybody build such a similar device?

/* HM-TC-IT-WM-W-EU
channeltype       channelnnr
WEATHER_TRANSMIT          1
THERMALCONTROL_TRANSMIT   2
WINDOW_SWITCH_RECEIVER    3
REMOTECONTROL_RECEIVER    6
SWITCH_TRANSMIT           7

WEATHER_TRANSMIT          1
TEMPERATURE               float     lesend über Ereignisse  -10, +50
HUMIDITY                  integer   lesend über Ereignisse  0   99

THERMALCONTROL_TRANSMIT   2
CONTROL_MODE              option    lesend über Ereignisse
LOWBAT_REPORTING          boolean   lesend über Ereignisse
COMMUNICATION_REPORTING   boolean   lesend über Ereignisse
WINDOW_OPEN_REPORTING     boolean   lesend über Ereignisse
BATTERY_STATE             float     lesend über Ereignisse
BOOST_STATE               integer
ACTUAL_TEMPERATURE        float
ACTUAL_HUMIDITY           float
SET_TEMPERATURE
AUTO_MODE
MANU_MODE
BOOST_MODE
COMFORT_MODE
LOWERING_MODE
PARTY_MODE_SUBMIT
PARTY_TEMPERATURE
PARTY_TEMPERATURE
PARTY_START_TIME
PARTY_START_DAY
PARTY_START_MONTH
PARTY_START_YEAR
PARTY_STOP_TIME
PARTY_STOP_DAY
PARTY_STOP_MONTH
PARTY_STOP_YEAR

WINDOW_SWITCH_RECEIVER    3
REMOTECONTROL_RECEIVER    6

SWITCH_TRANSMIT           7
DECISION_VALUE            integer   lesend über Ereignisse  0..255

knopserl, 2018-03-01:
thanks for the reply, even if it's dissapointing. I have now purchases many (30 :-) of the STM32 modules ato build a couple of thermostats hoping that I can do it with the Asksin++ library but it seem that I run into the next problem. SO what is required to build an own thermostat which has similar funktionality as the the one from Homematic? What need to be done to supporz the List7 to 9? Is there a chance to get it and if not what need to be done and what hints do I need or where/what do I need to change and how much effort would the mean?
Thanks and regards

knopserl, 2018-03-03:
Another set of question: Beside the request to extent the Asksin++ library to support List7 to 9 (I don't know how much effort that would be and if I can expect that):
Why would the additioonal Lists be required? Could I not use the List 0, 1, 3, 4 for the measurement values like: Temp, Humidity and Pressure and also for the commands the Thermostat would send? Why are the other Lists required? Are the Lists reserved or deticated? Is there any documentation what Lists can be used for what and what the limitations of the specific Lists are (a rule/guidance how to use the lists)? Or must the specific lists and names used exactly like the original Homematic Thermostat in order to work together with othe Homematic devices controlled by the Thermostat?

pa-pa, 2018-03-07:
As far as I understand are these lists for the thermostat programming. You can see the list descriptions in the XML files for the [CCU](https://github.com/eq-3/occu/tree/master/firmware/rftypes).

closed 2018-05-11 by pa-pa