pa-pa/AskSinPP #30: add new sensors

jp112sdl, 2018-03-05:
Hi papa,
kannst du bitte die beiden Sensor-Files für DS18B20 und BH1750 mit aufnehmen?
Ich musste die .h nach .txt umbenennen, um sie hier anhängen zu können...

Danke!
[Bh1750.txt](https://github.com/pa-pa/AskSinPP/files/1782285/Bh1750.txt)
[Ds18b20.txt](https://github.com/pa-pa/AskSinPP/files/1782286/Ds18b20.txt)

jp112sdl, 2018-03-05:
Überarbeitete Bh1750
[Bh1750.txt](https://github.com/pa-pa/AskSinPP/files/1782537/Bh1750.txt)

pa-pa, 2018-03-06:
Nochmal angepasst um dem Mode zu unterstützen. Die Fehlernummer konnte ich nicht nachvollziehen. Present kann vom begin-Returnwert übernommen werden. Bitte mal testen, bevor ich einchecke

```
// https://github.com/claws/BH1750
template <byte ADDRESS=0x23,::BH1750::Mode MODE=::BH1750::Mode::CONTINUOUS_HIGH_RES_MODE>
class Bh1750 : public Brightness {
  ::BH1750   _bh;
public:
  Bh1750 () : _bh(ADDRESS) {}
  void init () {
      _present = _bh.begin(MODE);
  }
  void measure (__attribute__((unused)) bool async=false) {
    if( present() == true ) {
      _brightness = _bh.readLightLevel();
    }
  }
};

```

jp112sdl, 2018-03-06:
Wie muss ich im Sketch jetzt den Sensor instanziieren?
`Bh1750< ??? >     bh1750;`

jp112sdl, 2018-03-06:
Ohne übergebene Parameter erhalte ich immer den Fehler:
`
.../sensors/Bh1750.h:12:43: error: 'BH1750::Mode' has not been declared
     template <byte ADDRESS=0x23,::BH1750::Mode MODE=::BH1750::Mode::CONTINUOUS_HIGH_RES_MODE>
`

pa-pa, 2018-03-06:
So

`  Bh1750<>          bh1750;
`

jp112sdl, 2018-03-07:
Das war auch mein erster Versuch. Aber immer wieder derselbe Fehler...
[image]

pa-pa, 2018-03-07:
Mach mal ein Update der Lib. Da gibt es jetzt ein Enum Mode, was er bei Dir och nicht kennt. Vielleicht geht aber auch uint8_t als Typ im Template.

jp112sdl, 2018-03-07:
Ah jetzt ja! :) 

Geht & läuft. Kannst einchecken 👍

closed 2018-03-07 by pa-pa