pa-pa/AskSinPP #33: RESET nicht möglich

jp112sdl, 2018-03-30:
Hi! :)

Seit Commit 923fe6b wird nach einem länger gedrücktem "longpressed" kein RESET mehr ausgelöst.

Vor 923fe6b:
[image]

Ab 923fe6b:
[image]

TomMajor, 2018-04-01:
kann ich bestätigen, mit aktuellem Stand auf Branch master kommt kein Reset nach longpressed mehr.

pa-pa, 2018-04-02:
War fälschlicherweise auskommentiert. Hatte mich gestört, da mein STM32 code immer einen Reset gemacht hat, wenn der Config-Button kurz gedrückt wurde :-(
Ist wieder geändert.

closed 2018-04-02 by pa-pa