pa-pa/AskSinPP #40: Dimmer for dualled

howy075, 2018-05-13:
Hello, could you perhaps integrate the actuator HM-LC-DW-WM?
greetings
Harald

pa-pa, 2018-05-15:
I currently have no plan to do it. May be if I have a lot of free time.

pa-pa, 2018-10-26:
HM-LC-DW-WM added for STM32

psi-4ward, 2019-03-03:
I've constructed a LC-DW with an Arduino and used `{0x01,0x08}` as Model.
The CCU shows it correctly but usage of the Channels is wrong.

**Actual:**
Chan 1 controls LED1
Chan 2 controls LED2

**Expected:**
Chan 1 controls the brightness
Chan 2 controls the color temperature

[image]

Another thing: I could not find/configure the on/off ramp times.

pa-pa, 2019-03-04:
Oh - the example is an HM-DW-WM - 2 channel dimmer. Can you please share your hardware setup and current code. The physical dimmer behavior can be changed by replace the DimmerControl class. Because the channels have different registers the channel configuration of the device needs also changed.

psi-4ward, 2019-03-04:
Thx for your fast answer.

The sketch is basically the same as your `HM-DW-WM.ino` expect the temperature and encoder stuff:

```cpp
#define EI_NOTEXTERNAL
#include <EnableInterrupt.h>
#include <AskSinPP.h>
#include <LowPower.h>

#include <Dimmer.h>
#include <DwDimmerChannel.h>

#define LED_PIN 4
#define CONFIG_BUTTON_PIN 8
#define DIMMER1_PIN 3
#define DIMMER2_PIN 5

#define PEERS_PER_CHANNEL 4
using namespace as;

// define all device properties
const struct DeviceInfo PROGMEM devinfo = {
    {0x01,0x08,0x01},       // Device ID
    "PsiDimDW01",           // Device Serial
    {0x01,0x08},            // Device Model (0x08: DualWhite; 0x09: 2-Ch)
    0x29C,                  // Firmware Version
    as::DeviceType::Dimmer, // Device Type
    {0x01,0x00}             // Info Bytes
};

typedef AvrSPI<10,11,12,13> SPIType;
typedef Radio<SPIType,2> RadioType;
typedef StatusLed<LED_PIN> LedType;
typedef AskSin<LedType,NoBattery,RadioType> HalType;
typedef DimmerChannel<HalType,PEERS_PER_CHANNEL> ChannelType;
typedef DimmerDevice<HalType,ChannelType,6,3> DimmerType;

HalType hal;
DimmerType sdev(devinfo,0x20);
DimmerControl<HalType,DimmerType,PWM8<> > control(sdev);
ConfigToggleButton<DimmerType> cfgBtn(sdev);

void setup () {
  DINIT(57600,ASKSIN_PLUS_PLUS_IDENTIFIER);
  if( control.init(hal,DIMMER1_PIN,DIMMER2_PIN) ) {
    sdev.channel(1).peer(cfgBtn.peer());
    sdev.channel(3).peer(cfgBtn.peer());
  }
  buttonISR(cfgBtn,CONFIG_BUTTON_PIN);
  sdev.initDone();
  sdev.led().invert(true);
  DDEVINFO(sdev);
}

void loop() {
  hal.runready();
  sdev.pollRadio();
}
```

And the Hardware is like https://asksinpp.de/assets/img/HM-LC-Dim1PWM.a394426a.svg
with a second MOSFET connected on PIN 5.

So we need to adjust both outputs every time a signal comes for Ch1 or Ch2:
Assumed Ch2 value is 50%. When we set the value of Ch1 to 100% both LEDs should be fully on.
If Ch2 value is 100% and Ch1 is 100% only the cold-white LED should glow.

Jerome pointed me to extend the `DimmerChannel` and add some custom `process()` implementations but I didn't try it yet.

pa-pa, 2019-03-04:
Can you please replace the DimmerControl by the new DualWhiteControl class in your code. This should use Ch1 for the brightness and Ch2 for the color adjustment.

psi-4ward, 2019-03-04:
It works! But probably we should better adjust the calculation to this to be able to power both LEDs with 100% at the same time:

[image]

[image]

pa-pa, 2019-03-04:
Better now ?

psi-4ward, 2019-03-04:
You're the mastmind! 
Works like a charm! Thx!! :+1: 

**For the record**  
It implements the linear crossfade with no pegel-limit.  
Cubic, Dim2Hot and Dim2Cold settings are ignored.

Commits: d4c06289ca3f6682fa00173f750f77b8922449fd, 01351065cfdf6a222e061a33f400cb8e5e4d4f10

psi-4ward, 2019-03-04:
Should we push the `HM-LC-DW_2Ch.ino` and `HM-LC-DW_DualWhite.ino` in the arduino version to the examples folder? I could prepare a PR for this.

pa-pa, 2019-03-04:
Yes, but we should use the "normal" HM names - HM-DW-WM & HM-LC-DW-WM

psi-4ward, 2019-03-04:
Hmm but the selling name is HM-DW-WM but the CCU lables it as HM-LC-DW-WM.
The original has a hardware switch to choose between 2-Ch and DualWhite mode.

pa-pa, 2019-03-04:
This strange - there are two different XML files

https://asksinpp.de/rftypes/rf_dim_2pwm.xml - 0x0109
https://asksinpp.de/rftypes/rf_dw.xml - 0x0108

psi-4ward, 2019-03-04:
interesting! so we can adopt the naming from there.
should I prepare a PR or would you just add the INO files?

pa-pa, 2019-03-05:
A PR would be nice :-)

psi-4ward, 2019-03-18:
#105

psi-4ward, 2019-08-21:
Oh think we could close this by #105

closed 2019-08-21 by pa-pa