pa-pa/AskSinPP #51: LongPress not correctly working for blind actor

Xento, 2018-07-24:
With my pullrequest https://github.com/pa-pa/AskSinPP/pull/48 some bugs are fixed but there is still a racecondition.
First: The original actor doesn't send statusupdates when moving.
 
This are the messages from an original blind actor when doing a long press and release after some time
>  -> 0B 43 84 40 3B1630 4AA8CC 43 09  - 2595907
-> 0B 44 84 40 3B1630 4AA8CC 43 09  - 2596173
-> 0B 45 84 40 3B1630 4AA8CC 43 09  - 2596440
-> 0B 46 84 40 3B1630 4AA8CC 43 09  - 2596706
-> 0B 47 84 40 3B1630 4AA8CC 43 09  - 2596974
-> 0B 48 84 40 3B1630 4AA8CC 43 09  - 2597240
-> 0B 49 84 40 3B1630 4AA8CC 43 09  - 2597507
-> 0B 4A 84 40 3B1630 4AA8CC 43 09  - 2597773
-> 0B 4B 84 40 3B1630 4AA8CC 43 09  - 2598039
-> 0B 4C 84 40 3B1630 4AA8CC 43 09  - 2598305
-> 0B 4D 84 40 3B1630 4AA8CC 43 09  - 2598572
-> 0B 4E 84 40 3B1630 4AA8CC 43 09  - 2598838
-> 0B 4F 84 40 3B1630 4AA8CC 43 09  - 2599104
-> 0B 50 84 40 3B1630 4AA8CC 43 09  - 2599370
-> 0B 51 84 40 3B1630 4AA8CC 43 09  - 2599636
ignore 0B 52 A0 40 3B1630 4AA8CC 43 09  - 2599903
ignore 0E 52 80 02 4AA8CC 3B1630 01 01 9C 20 47  - 2600030
ignore 0D 53 A4 10 4AA8CC 473071 06 01 9A 00  - 2602391
ignore 0A 53 80 02 473071 4AA8CC 00  - 2602514

Because your code sends periodical statusupdates some longpress messages are missed or are processed to late. You can see that the longpress works until a statusupdate is send and than the longpress is interrupted because the actor switched to the next state.
You can get around this problem when you set a higher value for maxTimeFirstDir but 0,5 seconds is the default and 0,8 minimal seconds when you use aes.

> 
-> 0B 4C 84 40 5667E7 5932AF 41 6A  - 140406
Switch from 03 to 01
Switch from 01 to 02
-> 0B 4D 84 40 5667E7 5932AF 41 6A  - 140673
Switch from 02 to 08
UP - 140679
<- 0E 36 A2 10 5932AF 473071 06 01 04 50 70  - 140793
-> 0A 36 80 02 473071 5932AF 00  - 140920
waitAck: 01
-> 0B 4E 84 40 5667E7 5932AF 41 6A  - 140939
-> 0B 4F 84 40 5667E7 5932AF 41 6A  - 141205
-> 0B 50 84 40 5667E7 5932AF 41 6A  - 141471
-> 0B 51 84 40 5667E7 5932AF 41 6A  - 141737
-> 0B 52 84 40 5667E7 5932AF 41 6A  - 142004
-> 0B 53 84 40 5667E7 5932AF 41 6A  - 142270
-> 0B 54 84 40 5667E7 5932AF 41 6A  - 142534
<- 0E 37 A2 10 5932AF 473071 06 01 0D 50 70  - 142704
-> 0B 55 84 40 5667E7 5932AF 41 6A  - 142809
-> 0A 37 80 02 473071 5932AF 00  - 142833
waitAck: 01
Switch from 08 to 03
STOP - 143034
-> 0B 56 84 40 5667E7 5932AF 41 6A  - 143067
Switch from 03 to 01
Switch from 01 to 02
-> 0B 57 84 40 5667E7 5932AF 41 6A  - 143333
Switch from 02 to 08
UP - 143339
<- 0E 38 A2 10 5932AF 473071 06 01 0D 50 6D  - 143454
-> 0A 38 80 02 473071 5932AF 00  - 143579
waitAck: 01
-> 0B 58 84 40 5667E7 5932AF 41 6A  - 143599
<- 0E 39 A2 10 5932AF 473071 06 01 0D 50 70  - 143722
-> 0A 39 80 02 473071 5932AF 00  - 143849
-> 0B 59 84 40 5667E7 5932AF 41 6A  - 143865
-> 0B 5A A0 40 5667E7 5932AF 41 6A  - 144132
<- 0E 5A 80 02 5932AF 5667E7 01 01 0D 50 6E  - 144252
Switch from 08 to 03
STOP - 144633
<- 0E 3A A2 10 5932AF 473071 06 01 11 00 6E  - 145362
-> 0A 3A 80 02 473071 5932AF 00  - 145489
waitAck: 01

Xento, 2018-07-24:
Maybe it could be a problem with the statusInfoMinDly in my case it is set to 2.
Maybe for 2 seconds per default from ccu2?
I can't configure the message delay or random time or the max retry count over the ccu. There seems to be some problem, too.

jp112sdl, 2018-07-24:
Ich schreibe jetzt hier mal auf deutsch...
In einem Vortrag von eQ3 auf Youtube meine ich gehört zu haben, dass der Rollladenaktor während der Fahrt keinen "Zwischenstand" sendet, sondern erst, wenn er anhält.

Hab nun gerade mal einen "echten" HM Rollladenaktor gesteuert und Pakete "mitgesnifft". 
Da kam zwischendrin nix... nur beim Stoppen.

Xento, 2018-07-25:
Ja genau.
Ich bin mir noch nicht ganz sicher woran es liegt.
Bei meinem Aktor ist der Wert von statusInfoMinDly auf 2 gesetzt, das entspricht dann nach der Berechnung im Code 1 Sekunde.

> void updateLevel (uint8_t l) {
    level = l;
    DDECLN(level);
    triggerChanged(decis2ticks(list1.statusInfoMinDly()*5));
  }

Entweder wird die Funktion updateLevel nicht häufig genug aufgerufen, damit der Alarm immer wieder zurückgesetzt wird oder aber die Statusnachricht sollte von einer anderen Stelle aus ausgelöst werden.

Ich glaube in der Klasse LevelUpdate sollte das Update nicht jede Sekunde, sondern z.B. alle 200ms ausgelöst werden.
Aufjedenfall sollte es unterhalb von 500ms liegen, da dies der geringste einstellbare Wert ist.

jp112sdl, 2018-07-25:
> Ich glaube in der Klasse LevelUpdate sollte das Update nicht jede Sekunde, sondern z.B. alle 200ms ausgelöst werden.

Ich hätte da Bedenken, dass es dann eher zu Kollisionen mit anderen Geräten/Paketen kommt.

Die Frage, die sich mir stellt:
Wofür gibt es die `updateLevel`-Methode?
Denn wie gesagt - die originalen Geräte senden _während der Fahrt_ keinen Level.

Das System muss ja auch noch funktionieren, wenn du bspw. 10 Aktoren zur selben Zeit fahren lässt. 
Es würde das reinste Funk-Chaos herrschen.
Das ist wohl auch der Gedanke von eQ-3, weshalb während der Fahrt keine Updates gesendet werden.
Das kann man am Ende der Fahrt machen.
Denn a) werden mit hoher Wahrscheinlichkeit nicht alle Rollläden im Haus die exakt selbe Fahrzeit haben und b) wenn es am Ende beim Übermitteln des Status zu einer Kollision kommt, wird er erneut übertragen.

Problematisch wäre es auch, wenn du einen Taster peerst, der ja beim LONG alle paar Millisekunden ein Event an den Aktor sendet. 

Die Option `statusInfoMinDly` dient dazu, nach dem STOPP den Ist-Level verzögert zu senden, um bei häufigen Wechseln (z.B.: man drückt auf den Runter-Taster, will zu einer bestimmten Position fahren, verpasst diese - STOPP - drückt wieder auf den Rauf-Taster fährt wieder etwas vorbei - STOPP - noch mal etwas runter usw.) das Funktelegramm nicht sofort zu senden, sondern erst nach der dort definierten Zeit, nach dem letzten STOPP.

Xento, 2018-07-25:
Das Problem tritt auf, wenn statusInfoMinDly auf 1 Sekunde oder niedriger gestellt wird.
Die CCU stellt es eigentlich auf 2 Sekunden ein und im Sketch sollte der Default auch 2 Sekunden sein.
Bei mir war der reine Werte aus dem Register aber 2, welches laut papas Umrechnung 1 Sekunde ergibt.

So jetzt zu dem Problem:
Der Alarm für das Statusupdate wird beim Aufruf der UpdateLevel Funktion durchgeführt.
Diese wird aber nur jede Sekunde aufgerufen, wenn der Alarm also kürzer oder gleich eingestellt ist, dann kann es vorkommen, dass er ausgelöst wird und dann das Update sendet.

Ich habe mal die LevelUpdate-Klasse angepasst so dass diese alle 200ms nen Update macht und siehe da es wird nichts mehr zwischendurch gesendet.
Die minimal per CCU einstellbare Verzögerung des Meldung ist 500ms also muss der Alarm von der Meldung vor ablauf der 500ms zurückgesetzt werden, damit keine Meldung geschickt wird.

> class LevelUpdate : public Alarm {
>     BlindStateMachine&  sm;
>     uint16_t  done;
>     uint16_t  fulltime;
>     uint8_t   startlevel;
>   public:
>     LevelUpdate (BlindStateMachine& m) : Alarm(0), sm(m), done(0), fulltime(0), startlevel(0) {}
>     ~LevelUpdate () {}
>     void start (AlarmClock& clock,uint16_t ft) {
>       startlevel = sm.status();
>       fulltime = ft;
>       done = 0;
>       set(decis2ticks(2));
>       clock.add(*this);
>       sm.changed = true;
>     }
>     virtual void trigger (AlarmClock& clock) {
>       // fulltime is 0.1s - so we add 10 for a second
>       done += 2;
>       if( done > fulltime ) done = fulltime;
>       uint8_t dx = (done * 200UL) / fulltime;
>       if( sm.state == AS_CM_JT_RAMPON ) {
>         if( dx > (200-startlevel) ) dx = 200-startlevel;
>         sm.updateLevel(startlevel + dx);
>       }
>       else if( sm.state == AS_CM_JT_RAMPOFF ) {
>         if( dx > startlevel ) dx = startlevel;
>         sm.updateLevel(startlevel - dx);
>       }
>       set(decis2ticks(2));
>       clock.add(*this);
>     }
>   };

Xento, 2018-07-25:
Jetzt hatte ich gerade nur noch ein Problem, dass eine Nachricht verpasst wurde .

Der Sender ist ein originaler 6-Fach Taster.
Bei nem Longpress sendet dieser alle 250ms dass der Button noch gedrückt ist.
maxTimeFirstDir ist auf 0.5 Sekunden gestellt, da ich in der CCU das Peering einfach als UP eingestellt habe. Wenn der Aktor AES aktiviert hätte, währe 0.8 Sekunden der geringste Wert und der Standardwert.

Ich habe mal die beiden Nachrichten groß gemacht.
Anhand des Zählers kann man sehen, dass scheinbar die Nachricht mit dem Zähler 00 verloren gegangen ist.
Daher wurde die maxTimeFirstDir-zeit erreicht und es wurde in den nächsten State gewechselt.
Bei den originalen Aktoren hatte ich noch nie das Verhalten, dass der Rolladen zwischendurch kurz stoppte ...
Haben die vielleicht bei nem Longpress ne größere Tolerzanz drin, wenn man eine Nachricht verloren geht?

> -> 0B F8 84 40 669723 5932AF 41 1E  - 123242
> Switch from 02 to 08
> UP - 123248
> <- 0E 09 A2 10 5932AF 473071 06 01 1D 50 5B  - 123363
> -> 0A 09 80 02 473071 5932AF 00  - 123490
> waitAck: 01
> 29
> -> 0B F9 84 40 669723 5932AF 41 1E  - 123508
> 30
> -> 0B FA 84 40 669723 5932AF 41 1E  - 123774
> 31
> -> 0B FB 84 40 669723 5932AF 41 1E  - 124041
> 32
> 33
> -> 0B FC 84 40 669723 5932AF 41 1E  - 124307
> 34
> -> 0B FD 84 40 669723 5932AF 41 1E  - 124571
> 35
> -> 0B FE 84 40 669723 5932AF 41 1E  - 124837
> 36
> 37
> ### -> 0B FF 84 40 669723 5932AF 41 1E  - 125104
> 38
> 39
> Switch from 08 to 03
> ### STOP - 125603
> ### -> 0B 01 84 40 669723 5932AF 41 1E  - 125634
> Switch from 03 to 01
> Switch from 01 to 02
> -> 0B 02 84 40 669723 5932AF 41 1E  - 125900
> Switch from 02 to 08
> btnOff - 125904
> UP - 125906
> <- 0E 0A A2 10 5932AF 473071 06 01 27 50 5A  - 126023

Xento, 2018-07-26:
Ich denke, dass der Fehler mit https://github.com/pa-pa/AskSinPP/pull/53 behoben wird.
Bei mir funktioniert nun der Longpress ohne Probleme.

pa-pa, 2018-07-30:
Ich habe das ganze nochmal etwas überarbeitet. Da waren auch noch Umrechnungsfehler drin. Bitte mal testen. Da der Default-Wert für CHANGE_OVER_DELAY angepasst wurde, ist ein RESET nach dem Update nötig.

pa-pa, 2018-07-31:
Oh man - war schon spät gestern. Nochmal die Regsiterwerte korregiert.

Xento, 2018-07-31:
Ich hab in meinem Pullrequest noch etwas an deinem neuen Code geändert.
Meiner Ansicht nach könnte unter Umständen beim Stoppen des LevelUpdate eine leicht andere Position, als durch den Trigger vorher gesetzt wurde, gesetzt werden.
Daher währe es vielleicht besser wenn man beim Auslösen des Triggers das LevelUpdate abbrechen würde und dann beim Stop in der SwitchState-Funktion sicherhstellt, dass das Level nicht überschrieben wird.

Ich bin noch nicht dazu gekommen das zu testen.

pa-pa, 2018-08-01:
Nein - das Trigger überschreibt in meiner aktuellen Implementierung garantiert das Level vom LevelUpdate-Alarm. Das ist so, weil der trigger() Aufruf der Basisklasse (StateMachine) jetzt vor dem Setzen des anzufahrenden DestLevel erfolgt. Durch die Abarbeitung innerhalb der StateMachine ist der LevelUpdate-Alarm schon gestoppt. Falls jetzt die Berechnung vom anzufahrenden Wert abweicht, wird durch das folgende updateLevel() das Level auf den anzufahrenden Wert gesetzt.

Xento, 2018-08-01:
Ah ok, jetzt hab ichs auch gesehen. War bei mir wahrscheinlich auch schon zu spät xD

Nen paar Kommentare währen vielleicht nicht schlecht im Code 👍

pa-pa, 2018-08-03:
Dann können wir das ja hier schliessen

closed 2018-08-03 by pa-pa