pa-pa/AskSinPP #52: Nicht alle Parameter sind per CCU konfigurierbar

Xento, 2018-07-26:
Ich habe jetzt schon mehrere Aktoren gebaut und bei keinem kann ich z.B. 

- Reset per Gerätetaste sperren
- Statusmeldungen Mindestverzögerung
- Statusmeldungen Zufallsanteil
- Max. Sendeversuche

Ich weiß, dass einige Funktionen nicht implementiert sind, aber ich würde doch sagen, dass sie konfigurierbar sein sollten.
Ich hab schon versucht mich schlau zu machen wodran die CCU erkennt welche Funktionen in der List1 angezeigt werden. Bin da aber leider noch nicht so weit gekommen.

jp112sdl, 2018-07-26:
Meinst du die Anzeige in der WebUI (die genannten Punkte fehlen gänzlich)?
Beziehst du dich auf emulierte "echte" HM Geräte oder eigene Device Models?

"Reset per Gerätetaste sperren" (DREG_LOCALRESETDISABLE) ist auf jeden Fall drin und geht.
Die Berücksichtigung der Anzahl der max. Sendeversuche (DREG_TRANSMITTRYMAX), ist seit kurzem implementiert.
Das sind aber alles Device-Register (List0).

Kanal-Register (List1) sind in der XML als Paramset direkt im Channel 1 abgelegt.

jp112sdl, 2018-07-26:
> Also DREG_TRANSMITTRYMAX und die Statusmeldungen sind List1 Einstellungen.

**D**REG_TRANSMITTRYMAX  ist in List0
**C**REG_TRANSMITTRYMAX  ist in List1

Der Blind-Typ hat die Option LOCALRESETDISABLE  erst ab Firmware >= 0x22.

Xento, 2018-07-26:
Ah ich habs jetzt gefunden ...
In der XML von meinem Device fehlen die Einstellungen.
Hatte die von ner älteren Version übernommen ...

closed 2018-07-26 by Xento