pa-pa/AskSinPP #58: Blind-Aktor fährt nicht korrekt auf Position wenn diese kleiner ist als die aktuelle

Xento, 2018-08-03:
Wenn man eine Position anfahren will per gepeertem Taster und diese kleiner ist als die aktuelle, dann fährt das Rollo trotzdem hoch.
Problem ist hier, dass es aktuell keine Möglichkeit gibt in einem solchen Fall in den State AS_CM_JT_RAMPOFF zu wechseln, da der Wert für die Position ja in SHORT_ON_LEVEL bzw. LONG_ON_LEVEL gespeichert ist.

Beispiel 1:
Position: 0%
SHORT_ON_LEVEL: 20%
Rollo fährt auf 20% (korrekt)

Beispiel 2:
Position: 40%
SHORT_ON_LEVEL: 20%
Rollo fährt auf 60% (falsch)

Es muss so umgebaut werden, dass das destlevel nicht beim getDelayForState aus dem Registern gelesen wird, da ja noch garnicht fest steht wohin gefahren werden muss.

Ich würde getNextState in der BlindStateMachine überschreiben und das Setzen des destlevel aus der 
Funktion getDelayForState entfernen.
Den Code habe ich noch nicht getestet. Vielleicht komme ich heute Nachmittag dazu ...

``` c
virtual uint8_t getNextState (uint8_t stat) {
    switch( stat ) {
      case AS_CM_JT_ONDELAY:  return AS_CM_JT_REFON;
      case AS_CM_JT_REFON:    
        destlevel = lst.onLevel();
        
        if(destlevel > level){
          return AS_CM_JT_RAMPOFF;
        }
        else {
          return AS_CM_JT_RAMPON;
        }
        
      case AS_CM_JT_RAMPON:   return AS_CM_JT_ON;
      case AS_CM_JT_ON:       return AS_CM_JT_OFFDELAY;
      case AS_CM_JT_OFFDELAY: return AS_CM_JT_REFOFF;
      case AS_CM_JT_REFOFF:   
        destlevel = lst.offLevel();
        
        if(destlevel < level){
          return AS_CM_JT_RAMPOFF;
        }
        else {
          return AS_CM_JT_RAMPON;
        }
      case AS_CM_JT_RAMPOFF:  return AS_CM_JT_OFF;
      case AS_CM_JT_OFF:      return AS_CM_JT_ONDELAY;
    }
    return AS_CM_JT_NONE;
  }
```

Xento, 2018-08-03:
Mal so nebenbei, müsste der nächste State eigentlich nicht immer über die Funktion getJumpTarget ermittelt werden?
In der CCU kann ich für jeden State einstellen welcher der nächste sein soll.
Darüber kann man z.B. nen Blinken eines Aktors erzeugen usw.

pa-pa, 2018-08-03:
Das mit der kleineren Position ist tricky. 
Ich würde da aber eher in der remote() Methode der BlindStateMachine (Blind.h Zeile 303) prüfen, ob ein die "normale" Richtung passt. Man kann ja bei ONDELAY/OFFDELAY in die Liste nach On/OffLevel prüfen. Falls die Richtung nicht passt, nutzt man einfach die setDestLevel() mit dem anzufahrenden Level.
Eine andere Methode wäre, direkt im Sketch bei switchState das aktuelle mit dem DestLevel vergleichen und dann den Motor einfach in die Gegenrichtung steuern.

pa-pa, 2018-08-03:
Bist Du Dir mit den States und der CCU sicher ? Ich denke, Du kannst nur für ein Peer einstellen, welcher State als nächstes angesprungen werden soll, in Abhängigkeit von aktuellen State. Dannach wird die Standardstatemachine abgearbeitet. Das blinken geht, da Du einfach eine Zeit für On und Off angibst. Der Aktor geht dann automatisch nach Ablauf der Onzeit in den OFFDELAY und entsprehcend nach Ablauf der Offzeit in den ONDELAY.

Xento, 2018-08-03:
Ja genau bei einem Peer kann man das einstellen.
Hast Recht mit den States, leider gibts dazu ja keine richtige Doku :-(
Dachte immer das man damit die Ausführungsreihenfolge der States beeinflussen kann.
Aber das stimmt nur, wenn ein zweiter Befehl innerhalb z.B. der OnDelay-Zeit rein kommt wohin dann gesprungen werden soll.

Die Variante mit dem switchState funktioniert so nicht, da dann die Berechnung der Position nicht mehr passt, da diese ja davon abhängt in welchem State also RampOn/Off man ist.
Also muss der richtige State gesetzt werden.

Ich versuch mal das ganze einzubauen.

Xento, 2018-08-03:
Wie du schon sagtest, ist das schon etwas tricky.
Ich zerbrech mir gerade den Kopf mit den States.

- es soll ON angefahren werden, ON ist 20%
- das Rollo steht aber auf 40%
- es muss also RampOFF gefahren werden
- wenn jetzt die standard Statemachine abgearbeitet wird, käme irgendwann der Status OFF anstatt ON
- OFF ist aber 100%

Ich versuch mal ob ich nen paar Sachen bei einem originalen Aktor herrausfinden kann.
Vielleicht muss die getNextState Methode auch virtual gemacht werden, damit man da auch eingreifen kann um z.B. nach dem RampON zu prüfen. ob vielleicht gerade die Position für OFF erreicht wurde um dann den State auf OFF zu setzen.

pa-pa, 2018-08-06:
Kannst DU das mal als Patch schicken / anhängen. Dann sieht man die Änderungen einfacher.

Xento, 2018-08-06:
Hier ist der Pullrequest: https://github.com/pa-pa/AskSinPP/pull/62
Vielleicht kann man das mit dem getNextState noch anders lösen, da ich so erstmal die Funktion stop rausnehmen musste, da ich von dort aus keine List mit den Einstellungen übergeben konnte.
Da das Stop ja aber von irgendwoher kommen sollte also von einem Peer ob intern oder extern, sollte man schon eine Liist zur Verfügung haben.

pa-pa, 2018-08-24:
PR merged

closed 2018-08-24 by pa-pa