pa-pa/AskSinPP #71: Trotz NACK wird weiterhin versucht die Message zu senden

Xento, 2018-08-30:
Ich bin darauf gestoßen, dass AskSinPP trotz NACK Antwort weiterhin versucht eine Nachricht zuzustellen.
https://github.com/jp112sdl/HB-UNI-Sen-WEA/issues/12

Sollte ein NACK nicht dafür sorgen, dass nicht erneut versucht wird zu senden?

pa-pa, 2018-08-30:
Derzeit wird nur auf Ack getestet. mach mal folgende Änderung in Device.h Zeile 248
`       result = response.isAck();
       if( response.isNack() ) maxsend = 0;
`

pa-pa, 2018-08-30:
Kann ich machen - aber ich würde gern vorher wissen, ob es auch den gewünschten Effekt hat.

pa-pa, 2018-08-31:
Ich habe es jetzt einfach mal eingebaut.

closed 2018-08-31 by pa-pa

Xento, 2018-09-01:
Danke ;-)
In meinem Fall wirds aufjedenfall helfen und ich denke, das generell nicht die gleiche Nachricht immer wieder gesendet werden soll, wenn diese bereits abgelehnt wurde.
Das Verzögert nur die weiteren Nachrichten und verursacht mehr Funkauslastung.