pa-pa/AskSinPP #72: nRF52 support

knopserl, 2018-09-03:
even if the ESP (ESP32, ESP8266) are not supported, you helped me a lot to get the ESP32 running, but I'm still stuck with ESP32 battery usage and the sleep/idle function.
The biggest issues with the ESP MCU is there wake-up behavior (soft restart).
Thus I have decided to do futher investigations with the very power efficient nRF52 based boards such as the nRF52832 boards below (I have ordered some).
The advantage of this chip/barebone boards that has ultra low power capabilities (esp. build for that) and it has a lot of peripheral support and it does continue the code after the wakeup (like the other MCU's).
As I still would like to build a battery powered Homematic room sensor with E-Paper display, I would require your support.
BTW: THe nRF52832 has 512kB Flash and 64kB RAM and thus it does not run out of memory like the ATMEGA328p or others.

ANother question: Does the STM32 support include now full sleep sleep support or do I have the same issue as with the ESP32 (no sleep thus no battery mode support)? 

https://de.aliexpress.com/item/nRF52832-2-4GHz-Wireless-rf-Module-CDSENET-E73-2G4M04S-SPI-SMD-rf-Receiver-transmitter-Bluetooth-Module/32819293925.html?spm=a2g0x.search0104.3.1.3b9652d4CsTYGo&ws_ab_test=searchweb0_0,searchweb201602_5_10065_10068_5014915_10547_10059_10548_10696_100031_10084_10083_10103_451_10618_5014815_452_10304_10307_10820_10821_10302,searchweb201603_1,ppcSwitch_3&algo_expid=d697b633-dade-447d-aa4c-9b0063c4aeb8-0&algo_pvid=d697b633-dade-447d-aa4c-9b0063c4aeb8&transAbTest=ae803_1&priceBeautifyAB=0

pa-pa, 2018-09-04:
STM32 has currently no sleep support. I use is only for devices with permanent power supply.

knopserl, 2018-09-04:
ok, thene it's not better than my ESP32 support (and I prefer the ESP32 over the STM32).

But for battery it would be great to have nRF52 support. If you have no plans to support it, I might start another attempt for this MCU (great for battery) and you might support me to do the port?

pa-pa, 2018-09-04:
What about ATMega644 or ATMega1284 or ATMega2560. All of these have more flash and ram. And they are already support.
If this will not work I will try to help you again.

knopserl, 2018-09-04:
I have not really found a convincing board which has about 20k RAM with ATMegas. Do you? The most convincing boards a nRF52 based (that is also what other communities recommended for battery sensors). They have 512k and 64k for about 4€

closed 2018-10-26 by pa-pa

knopserl, 2018-12-06:
@knopserl. Iknow this is already closed, but a need a hort answer from you:
I'll give the nRF52 (840) a try. It seem to be perfect for battery usage and does not restart after deep sleep. So I'll start another port.
The nRF52 does not have a real EEPROM but there is a EEPROM emulation using flash, thus I would like to know what the number of write cycles is expected under "normal" conditions? Is it a challange for the flash are EEPROM writes only required for pairing and configuration changes and thus not that often like restarts or even during normmals with sleep/wakeup?

pa-pa, 2018-12-07:
This should no problem. The EEPROM is onölyused for settings. This means it is only written during pairing and when new register settings are send. See the STM32 port - it also uses the last 1kB flash for saving the register when no external EEPROM is used. See InternalEprom class in Storage.h

knopserl, 2018-12-07:
@pa-pa Thanks for the fast answer: I have a EEPROM emulation and a LowPower class for the nRF52, so I hope the port for it will be managable. I might need a few hints when I get stuck, but I do my best not disturbing you often.