pa-pa/AskSinPP #80: pwmtable starts at 1?

psi-4ward, 2018-10-29:
I've drafted the Dim1PWM 

[image]

And uploaded `HM-LC-Dim1PWM-CV.ino`

* LED Strip is off on power on
* Dimming between 10-100% works as expected
BUT when I dimm to 0% the LEDs are still glowing cause the GPIO still emits.

When I alter the [pwmtable](https://github.com/pa-pa/AskSinPP/blob/master/actors/PWM.h#L15) to start from 0 it works as expected.

Is this a Bug or am I doing smth wrong?

pa-pa, 2018-10-29:
This was a bug. Code changed to ensure LED off when value is 0%.

closed 2018-10-29 by pa-pa

psi-4ward, 2018-10-29:
Thanks for your fast answer! 
But the changes aren't pushed yet, are they?

pa-pa, 2018-10-29:
You are right. They are pushed now.

psi-4ward, 2018-10-29:
2871c681c0735f2dc2babd11386586903035135d

Perfect thanks.