pa-pa/AskSinPP #86: Dimmer mit 2 Buttons "komisches verhalten"

psi-4ward, 2018-11-21:
Die _internen Gerätetasten_ verhalten sich etwas merkwürdig.

Nach dem Anlernen sind beide Tasten auf Experte eingestellt. Kann man hier `Dimmer aus / dunkler` bzw `Dimmer ein / heller` als Standard hinterlegen?

**Verhalten**
Drücke ich kurz auf Btn1 geht die LED an.
Drücke ich kurz auf Btn2 geht die LED aus.
Drücke ich lang auf Btn1 dimmt die LED hoch.
Drücke ich lang auf Btn2 dimmt die LED runter.

Änder ich in der CCU die Config der Buttons zB auf 
Btn1: Dimmer an; Pegel 50%
Btn2: Dimmer aus

Dann muss ich den Btn1 zweimal drücken damit die LED auf 50% geht.
Auch Btn2 schaltet die LED erst nach dem zweiten press aus.
Ich sehe jeweils beide "pressed" im Monitor.

Ändere ich nun den Dimmwert per CCU, dann schaltet Btn2 die LED nicht aus.
Bei einem LONGPRESS dimmt die LED runter. Jetzt geht ein SHORT wieder zum Ausschalten.

<details>
  <summary>Sketch</summary>
<p>

```cpp
//- -----------------------------------------------------------------------------------------------------------------------
// AskSin++
// 2016-10-31 papa Creative Commons - http://creativecommons.org/licenses/by-nc-sa/3.0/de/
//- -----------------------------------------------------------------------------------------------------------------------

// define this to read the device id, serial and device type from bootloader section
// #define USE_OTA_BOOTLOADER

#define EI_NOTEXTERNAL
#include <EnableInterrupt.h>
#include <AskSinPP.h>
#include <LowPower.h>

#include <Dimmer.h>

// we use a Pro Mini
// Arduino pin for the LED
// D4 == PIN 4 on Pro Mini
#define LED_PIN 4
// Arduino pin for the config button
// B0 == PIN 8 on Pro Mini
#define CONFIG_BUTTON_PIN 8

#define DIMMER_PIN 3
#define BUTTON1_PIN 7
#define BUTTON2_PIN 9

// number of available peers per channel
#define PEERS_PER_CHANNEL 4

// all library classes are placed in the namespace 'as'
using namespace as;

// define all device properties
const struct DeviceInfo PROGMEM devinfo = {
    {0x11,0x12,0x22},       // Device ID
    "papa111222",           // Device Serial
    {0x00,0x67},            // Device Model
    0x25,                   // Firmware Version
    as::DeviceType::Dimmer, // Device Type
    {0x01,0x00}             // Info Bytes
};

/**
 * Configure the used hardware
 */
typedef AvrSPI<10,11,12,13> SPIType;
typedef Radio<SPIType,2> RadioType;
typedef StatusLed<LED_PIN> LedType;
typedef AskSin<LedType,NoBattery,RadioType> HalType;
typedef DimmerChannel<HalType,PEERS_PER_CHANNEL> ChannelType;
typedef DimmerDevice<HalType,ChannelType,3,3,PWM8<> > DimmerType;

HalType hal;
DimmerType sdev(devinfo,0x20);
ConfigButton<DimmerType> cfgBtn(sdev);
InternalButton<DimmerType> btn1(sdev, 4);
InternalButton<DimmerType> btn2(sdev, 5);

void setup () {
  DINIT(57600,ASKSIN_PLUS_PLUS_IDENTIFIER);
  if( sdev.init(hal,DIMMER_PIN) ) {
    sdev.channel(1).peer(btn1.peer(), btn2.peer());
  }
  buttonISR(cfgBtn, CONFIG_BUTTON_PIN);
  buttonISR(btn1, BUTTON1_PIN);
  buttonISR(btn2, BUTTON2_PIN);
  sdev.initDone();
}

void loop() {
  hal.runready();
  sdev.pollRadio();
}
```
</p>
</details>

pa-pa, 2018-11-29:
Die Verarbeitung des Messagecounters war fehlerhat. Er wurde auch bei Short-Press ausgewertet. Dadurch wurden Messages von unterschiedlichen Quellen, aber mit gleichen Counter ignoriert. Bitte mal ausprobieren.

psi-4ward, 2018-11-29:
Vielen Dank!
Mein Flasher ist leider kaputt, wird paar Tage dauern.

psi-4ward, 2018-12-21:
fetzt! Danke

closed 2018-12-21 by psi-4ward