pa-pa/AskSinPP #102: implement RFM69 support?

mamama1, 2019-02-05:
Hi,

very great work here, Thank you so much for your effort. I'm extremely excited to jump into this and start developing my own Homematic devices. In this process I hope to learn more about this library, the AskSin protocol and so forth to hopefully eventually be able to contribute something to your great work.

One thing however, made me curious. Would it be possible for you to start working on RFM69 support at some point? Those RFM69 modules are much easier to source (e.g. from official distributors like Pollin or Mouser) and seem to have much better range and TX power.

I think, because of this, it would be a logical step to make this library work with the RFM69 modules, to be future-proof.

There has already been some effort here:
https://forum.fhem.de/index.php?topic=49300.0
https://forum.fhem.de/index.php?topic=57486.660
https://forum.fhem.de/index.php?topic=48433.0

but unfortunately at the moment I really barely understand what's going on in those threads - firstly because I'm not that experienced in RF technology (no idea what DataWhitening or preambles are, e.g.)  and secondly because the whole Homematic protocol is totally new for me and I have absolutely no idea what is going on on low level tx/rx layer.

So, once again, what is your opinion on implementing RFM69 support?

pa-pa, 2019-02-06:
I would be happy if we could support RFM69 too, but I have no time to do this.
If you want to take a look into this, you may copy the Radio class and start replacing all functions accessing the SPI - because they are implement the hardware behavior.

mamama1, 2019-02-06:
Time is a really big issue on this end as well.
However, I will try. I saw that you were very helpful in other threads/issues when people were trying to implement things. I hope you'll try to help me as well, when I eventually get to trying to implement the RFM69 module :-)

pa-pa, 2019-02-06:
I will try. May be it will be a good start to split the Radio class first in Radio & CC1101 parts. So the RFM69 need only to replace the CC1101 part. This can be later specified by a temlpate argument.

mamama1, 2019-02-06:
maybe you could start with that (separating the CC1101 part) and I'll later try to implement RFM69?

pa-pa, 2019-02-07:
So my part is done. Now it's your turn.

mamama1, 2019-02-07:
cool, thanks! I'll do my best, however I'll certainly need more time than you. I'll keep you posted.

pa-pa, 2020-09-08:
no progress - close