pa-pa/AskSinPP #130: Config Freq nach RESET gelöscht?

psi-4ward, 2019-08-25:
Bootlog **nachdem** zuvor FreqTest ausgeführt wurde:

```
AskSin++ V4.1.0 (Aug 26 2019 00:17:46)
Address Space: 32 - 1650
CC init1
CC Version: 14
 - ready
Config Freq: 0x21657A
ID: 010801  Serial: papa010801 debounce
 pressed
 longpressed
 longpressed
RESET
AskSin++ V4.1.0 (Aug 26 2019 00:17:46)
Address Space: 32 - 1650
00000000
Init Storage: CAFE6963
CC init1
CC Version: 14
 - ready
ID: 010801  Serial: papa010801
```

Nach dem `RESET` wird `Config Freq: 0x21657A` nicht mehr ausgegeben. Bedeutet dies, dass die Freqenzanpassung gelöscht ist? Afair sollte das in der Config-Area gespeichert sein und einen RESET überleben?

psi-4ward, 2019-08-26:
Gibt es für dich Sinn, dass ein Adress-Space-Overflow den Config Bereich 0-28 beschädigt/überschreibt?

Flashe ich mit 2 Peers/Chan bekomme ich _Address Space: 32 - 870_ und nach einem RESET bleibt sc.valid() true und die Config-Freq wird wie erwartet ausgegeben.

pa-pa, 2019-08-26:
Er berechnet wahrscheinlich irgendeine Mondchecksumme für den nicht existierenden Speicher. Das passt dann nicht mit der gespeicherten überein und als Fallback wird erst mal alles neu initialisiert :-)

PR #131 (2019-08-26): Reduce PEERS_PER_CHANNEL in DW-WM Sketches
4 Peers allocs address space 32 - 1650 but only 1kb is available
Closes #130
commit e9a0c519 (2019-08-26): Reduce PEERS_PER_CHANNEL in DW-WM Sketches

4 Peers allocs address space 32 - 1650 but only 1kb is available
Closes #130
closed 2019-08-26 by pa-pa