pa-pa/AskSinPP #133: HM-SEN-MDIR-WM55 mit zusätzlichem Relais

fraeggle, 2019-09-25:
Zu aller erst......  Super Projekt. 
Dies ist kein Fehler, sondern ne Anfrage.
Verwende seither ESP8266 mit ESPEasy, IOBroker und CCU um meine "Wünsche" der Hausautomation zu realisieren, aber so wären manche Dinge wahrscheinlich stabiler.
Ich wollte fragen ob es möglich wäre dem HM-SEN-MDIR-WM55 zusätzlich zu den Tasten die in dem Sketch sind auch ein Relais Port mit ein zu binden?
Habe aktuell einen ESP mit PIR, BH1750, Taster und Relais, da aber komischerweise an dieser Stelle Wlan deutlich schlecht und der 868Mhz Empfang der CCU mit 70db deutlich stabiler ist, wäre es dann eine Alternative. Leider hab ich noch nicht so richtig den Code kapiert (was zum erweitern nötig wäre).

Gruß Peter

jp112sdl, 2019-09-25:
Unabhängig vom Code müsstest du auch in der CCU ein neues Gerät implementieren.
Der HM-SEN-MDIR-WM55 hat ja von Hause aus keinen Aktorkanal.

fraeggle, 2019-09-25:
Ahh ok.. ich dachte das macht dann der "Sketch" automatisch wenn ich das an die CCU anlerne.
Weil in diesem Sketch ja z.B. auch 2 zusätzlich Tasten drin sind.
ich ging praktisch davon aus, dass dies mit typedef dann geschieht....

jp112sdl, 2019-09-25:
>  ich dachte das macht dann der "Sketch" automatisch wenn ich das an die CCU anlerne.

Nein, sowas geht nicht. 

> Weil in diesem Sketch ja z.B. auch 2 zusätzlich Tasten drin sind.

Was meinst du mit "zusätzlich 2 Tasten" ?
Der HM-SEN-MDIR-WM55 ist nativ ein 2fach Taster mit Bewegungsmelder.
https://de.elv.com/elv-homematic-komplettbausatz-funk-bewegungsmelder-mit-taster-fuer-55er-rahmen-hm-sen-mdir-wm55-fuer-smart-home-hausautomation-140848