pa-pa/AskSinPP #143: HM-LC-SWX-SMX

fraeggle, 2019-10-25:
Frage dazu.
Ich hab den Sketch geladen.
Habe folgendes im Sketch eingestellt:

// number of relays by defining the device
//#define HM_LC_SW1_SM 0x00,0x02
//#define HM_LC_SW2_SM 0x00,0x0a
#define HM_LC_SW4_SM 0x00,0x03

#define HM_SENSOR_RELAY
dann müsste doch der erste Part "angesprochen" werden.

#ifdef HM_SENSOR_RELAY
  // relay pins for the HMSensor Relay Board
  #define RELAY1_PIN 17
  #define RELAY2_PIN 16
  #define RELAY3_PIN 15
  #define RELAY4_PIN 14
  #define BUTTON1_PIN 6
  #define BUTTON2_PIN 3
  #define BUTTON3_PIN 19
  #define BUTTON4_PIN 18
#else
  // relay output pins compatible to the HM_Relay project
  #define RELAY1_PIN 5
  #define RELAY2_PIN 6
  #define RELAY3_PIN 7
  #define RELAY4_PIN 3
#endif

in HM erkennt er das Modul

[image]
Die Ausgänge schalten sauber über HM.
Nur dachte ich, dass ich dann mit den 4 Buttons die entsprechenden Ausgänge schalten kann, Aber da passiert nichts.
Oder bin ich da falsch. Acho Taster hab ich als schließer gegen GND an die PINS 3,6,18,19
Gruß Peter

jp112sdl, 2019-10-25:
Mach mal einen RESET. Configtaster gedrückt halten bis RESET seriell ausgegeben wird. 
Manchmal werden beim ersten Mal die internen Peerings nicht angelegt.

fraeggle, 2019-10-25:
du bist sowas von genial  :-)
Funktioniert  Mein erstes selbst gebautes Projekt von dir ..... Sowas von cool  dann kann ich meine ESP demnächst tauschen.
Danke nochmal  Gruß Peter.

Achso ich schließ dann gleich das Ticket

closed 2019-10-25 by fraeggle

jp112sdl, 2019-10-25:
Der Dank für den Sketch gilt @pa-pa :)

Ich hab hier nur eine Idee zur Problembehebung kommentiert. :)

Aber freut mich, dass das die Lösung war.