pa-pa/AskSinPP #154: Low-Bat.-Schwelle nur 5-15Volt

chkbln, 2019-11-23:
Hallo
ich nutze HM-LC-SW1-BA-PCB.ino und möchte die Hardware mit 2 Batterien betreiben.
also mit 3,3 Volt und niedriger. In der CCU2 lässt sich die Low-Bat.-Schwelle nur im Bereich 5-15Volt einstellen. Dadurch habe ich eine Fehlermeldung in der CCU und kann keine sinnvolle Warnung unter 3,3 Volt einstellen. Gibt es eine Möglichkeit das zu ändern? Wenn ja was müsste ich ändern?
Gruß
Christof

jp112sdl, 2019-11-23:
Hi, schau mal hier:
https://homematic-forum.de/forum/viewtopic.php?f=76&t=54517#p542850

Einfach in der WebUI einen mit 3 multiplizierten Wert eingeben (bspw. 9V) und dann im Sketch / 3 dividieren, um 3V abzubilden.

chkbln, 2019-11-24:
Hi
danke für Info. Das hat funktioniert.

Gruß
Christof

pa-pa, 2019-11-26:
not really an issue

closed 2019-11-26 by pa-pa