pa-pa/AskSinPP #155: Dimmer: 'WORKING' state wird bei "heller/dunkler"-Steuerung mittels internem Peering nicht zurückgesetzt

jp112sdl, 2019-11-25:
Wird über das interne Peering (ConfigToggleButton) das Dimmlevel schrittweise erhöht/verringert, bleibt der Status "WORKING" in der CCU stehen.

Was ich bisher erfassen konnte:
**Steuerung über CCU:**
```

Empfange 5% Dimmlevel
-> 0E 0F A0 11 4A80E8 010801 02 01 0A 00 A0  - 12451
Ramp/Level: 50/10
Bestätige mit WORKING
<- 0F 0F 80 02 010801 4A80E8 01 01 00 50 24 00  - 12580
Beende WORKING
<- 0F 07 A2 10 010801 4A80E8 06 01 0A 00 24 0A  - 15060
Erhalte Quittung
-> 0A 07 80 02 4A80E8 010801 00  - 15185
waitAck: 01
```
_-> finaler Status ist korrekt._

**Steuerung über Taster, Funktion: "Dimmer heller":**
```
 debounce
 pressed
 released
-> 0B 08 02 40 010801 010801 01 08  - 88344

Neuer Dimmlevel wird an CCU gesendet inkl WORKING
<- 0F 09 A2 10 010801 4A80E8 06 01 14 20 26 14  - 90370

-> 0A 09 80 02 4A80E8 010801 00  - 90494
waitAck: 01
```
_-> WORKING bleibt bestehen._

**Steuerung über Taster, Funktion: "Dimmer ein":**
```
 debounce
 pressed
 released
-> 0B 10 02 40 010801 010801 01 10  - 343898
Ramp/Level: 50/200

Neuer Dimmlevel wird an CCU gesendet; alles i.O. - kein WORKING State
<- 0F 11 A2 10 010801 4A80E8 06 01 C8 00 28 C8  - 346529

-> 0A 11 80 02 4A80E8 010801 00  - 346654
waitAck: 01
```
_-> finaler Status ist korrekt._

stan23, 2021-09-20:
Dieses Problem besteht immer noch:
in der CCU wird das Zahnrad angezeigt.

stan23, 2021-10-16:
Der Grund ist dass bei der lokalen Bedienung der Wert `alarm.destlevel` nicht aktualisiert wird, und deswegen bei der Berechnung der Flags immer ein Delta erkannt wird:
https://github.com/pa-pa/AskSinPP/blob/master/Dimmer.h#L678L682

PR #286 (2021-10-16): don't set WORKING if brightness was changed locally
fixes #155

`alarm.destlevel` keeps the previous value if the brightness is changed locally on the device, so the flags were calculated in a wrong way.
Fix: check `alarm.destlevel` only if an alarm is active.
commit f446b7a9 (2022-02-12): don't set WORKING if brightness was changed locally

fixes #155
closed 2021-10-16 by pa-pa