pa-pa/AskSinPP #178: Kanalparametern ohne Wirkung ?

weissjuergen, 2020-03-29:
Ich hab mir eine Schaltung für den Bewegungsmelder zusammengebaut und den entsprechenden Sketch draufgespielt. Funktioniert auch als Bewegungsmelder prima, nur leider sehr empfindlich.
Ich habe dann versucht bei den Kanalparametern Anzahl Impulse pro Zeit hochzustehen. Allerdings habe ich den Eindruck, dass das Einstellen von Kanalparametern keine Wirkung hat. Werden diese unterstützt ?

pa-pa, 2020-03-29:
Keiner der Filter-Parameter ist unterstützt.

weissjuergen, 2020-03-30:
Ich bin ja bin diese Projekten und Sketchen ein Neuling. Läßt sich diese Unterstützung programmieren oder geht das mit einem Arduino gar nicht und bleibt nur der Original Hardware von HM vorbehalten ?

pa-pa, 2020-03-30:
Das wird sicherlich gehen. Aber ich habe keine Ahnung, wie das genau funktionieren soll und brauche es auch nicht. Helfe aber gern, wenn sich jemand dran macht.

weissjuergen, 2020-03-31:
Wenn du das auch nicht weisst, gebe ich als Einsteiger erst mal auf.

closed 2020-03-31 by weissjuergen

jackie-t, 2020-11-29:
Was schon ein bisserl abschreckend ist wenn derjenige, _der AskSinPP laut Source komplett geschrieben hat_, sagt "boah, keine Ahnung wie das gehen soll".

Vielleicht wäre es wichtig, mal Dokumentation zu schreiben, dann könnten auch andere mehr mithelfen. So ist die Einstiegshürde - sogar für jemanden vom Fach - extrem hoch. Nicht weil ich nicht programmieren könnte, sondern weil ich mich durch eine Vielzahl von Klassen lesen müsste, um das "Big Picture" zu verstehen. Und das könnte mit einer Dokumentation erleichtert werden.

Wenn ich mir zum Beispiel HM-SEC-MDIR.ino ansehe, dann sticht gleich mals ein Hardware Abstraction Layer ins Auge, mit einem BaseHal, garniert mit Templates links und rechts. Scrolle ich runter, sehe ich die loop, mit "runready" und "pollradio", sonst nichts. Jetzt nehme ich mal an dass runready mit dem Senden zu tun hat, da gibt es aber auch ein pollradio, auf dem sdev, welches - wenn ich nach oben scrolle - ein MotionType ist, welcher wiederum per typedef eigentlich ein MultiChannelDevice ist. Bumm. Spätestens da wird es klar, dass ich mir nicht nur ein Verständnis von deinem Code erarbeiten müsste, sondern auch ein tiefgreifendes Verständnis der Struktur von Homematic-Devices mit ihren Channels benötigen werde. Also wenn wir gerade mal von Einstiegshürde werden.

Jetzt ignoriere ich das alles mal, und schau einfach weiter. Das MultiChannelDevice wird mit dem MChannel instantiert, welches im Endeffekt ein MotionChannel ist, mit dem Hal (was macht der nochmals?), den PEERS_PER_CHANNEL (da müsste man die Homematic Interna verstehen), die List0 (was ist das nochmals schnell), und dem Gerät. MotionChannel wird wohl im Motion.h sein, also geht's dort weiter.

Im Motion.h geht's gleich mit dem DEFREGISTER Marko los (Homematic Interna?), die werden instantiert, nehme mal an das sind die gefragten Einstellungen, also weiter. MotionEventMsg wird wohl die Nachricht sein, die ich nicht anrühren müsste, also ignorieren. Der MotionChannel, der als nächstes kommt, klingt interessant, das wird's wohl sein. Gleich mal mit Mehrfachvererbung von Channel und Alarm, ignoriere das aktuell.
Dann gibt's die QuietMode, keine Doku, aber nehme mal an mit dem MinInterval das ist die Verzögerung damit er nicht zweimal in X Sekunden auslöst. Warum trotzdem motiondetected() aufgerufen wird verstehe ich in dem Kontext nicht, egal. Weiter.

Die Klasse selbst scheint eine heiße Spur zu sein, weil in setup() geht's los. pirInterruptOn() wird aufgerufen - fein, da wird die ISR reingehängt. Aber nein, stattdessen wird nur eine Variable auf true gesetzt, die nach dem Standardkonstruktor bereits true ist. Ich scrolle weiter runter um zu sehen, wo jetzt die ISR ist, stoße auf trigger. Wer das aufruft ist mir egal, ich will nur mal weiterschauen, denn jetzt geht's los.

Als erstes wird cycle gecancelled, wegen "reset state timer because motion will be send now" - was war cycle nochmals? Raufgescrollt - aha, das ist die 5-Minuten Helligkeitsmeldung. Auch nicht besonders sprechend der Name, aber egal. Verstehe, wir brauchen die Helligkeit für die nächsten 5 Minuten nicht senden, weil jetzt gesendet wird. Da werden die Codeanpassungen wohl hingehören.

Was überhaupt nicht hilft ist, dass Tabs und Spaces gemischt sind, so ist der Code auf Github schwer zu lesen, weil die Einrückung inkonsistent ist. Schön langsam verstehe ich dann den Code - hier wird geschaut ob quiet aktiv ist, und wenn ja, dann wird am Ende nochmals ein Event ausgelöst (nehme mal an das ist bei Homematic so?). Das ganze ist eine Klasse um mit Alarm später das Ende detektieren zu können. Und wenn ich jetzt weiter runterscrolle sehe ich auch die ISR, in einem riesigen, mehrzeiligen C Makro mit Templates, und mehr inkonsistentem Einrücken und Code-Smells (Vergleiche auf == true), weswegen ich mir das erspare.

Im Endeffekt weiß ich jetzt, wo ich das ganze ändern muss. Und diese ganzen Schritte könnte man aber ersparen, wenn man mal das "Big Picture" des Codes irgendwo hinschreibt, wo man es leicht findet. Also wie hat der Code zu funktionieren, wo stehen die Klassen, usw. Schon mal zu sagen "OK, prinzipiell funktioniert alles mit Alarms, die mit sysclock eingehängt werden, und deren Ausführung mit runready (vielleicht? das ist jetzt nur eine Annahme, weil ich mir den Rest des Codes nicht durchgelesen habe) gestartet wird wenn's Zeit dafür ist -- all das würde schon immens helfen, und potentiellen Helfern Zeit ersparen. Wenn jetzt noch die ChannelDevices erklärt werden, und wie das mit den ISR gemeint ist damit ich nicht die Makros auflösen muss, dann wird's noch einladender, und somit die Einstiegshürde, die aktuell noch sehr hoch ist, niedriger.

Im Endeffekt ist das aber hier alles für die Katz, denn die meisten PIR Sensoren haben einen 5-6 Sekunden langen Cooldown, bevor sie wieder auslösen, und eignen sich nicht für Auswertungen wie "2 Signale in 1 Sekunde", weil man nie zwei Signale in einer Sekunde bekommen wird. Vielleicht geht's mit anderer Hardware, aber die ganzen geläufigen billigen Sensoren scheinen alle diese Einschränkung zu haben. Und _deshalb_ wird's schwierig.

jp112sdl, 2020-11-29:
> Was schon ein bisserl abschreckend ist wenn derjenige, der AskSinPP laut Source komplett geschrieben hat, sagt "boah, keine Ahnung wie das gehen soll".

Bei Hobby-Projekten ist es doch meist so, dass man sich selbst was baut/programmiert, wie es für den _eigenen_ Zweck notwendig ist. Man arbeitet sich in das Thema ein, bis es dann irgendwann funktioniert.
Aus gutem Willen denkt man sich: Da könnten auch andere Spaß dran haben - ich stelle meinen Code mal hier frei zur Verfügung. Ganz unverbindlich und kostenlos.

> Vielleicht wäre es wichtig, mal Dokumentation zu schreiben, dann könnten auch andere mehr mithelfen. So ist die Einstiegshürde - sogar für jemanden vom Fach - extrem hoch. Nicht weil ich nicht [...]

Dokumentation ist immer ein leidiges Thema. Und selbst wenn man alles haarklein aufschreibt - man kann sich sicher sein, dass wieder einer kommt, dem wieder etwas unklar ist und man erklärt wieder von vorn.
Unter `/doc` ist ja schon ein bisschen was Doxygen-Erstelltes zu finden.

Letztlich hat pa-pa die AskSin/NewAskSin von [trilu](https://github.com/trilu2000) "umverpackt" und erweitert, ohne wohl jedes Detail aus der Homematic-Welt zu kennen (z.B. wie genau der Eventfilter zu funktionieren hat).

Dazu kamen dann im Laufe der Zeit auch weitere Änderungen und Erweiterungen durch etliche [Contributoren](https://github.com/pa-pa/AskSinPP/graphs/contributors)

Selbst ich (mit Nicht-Programmierer Vergangenheit) habe über 50 Geräte (Original-Nachbauten sowie eigene Typen) erstellt, ganz ohne "Big Picture". Habe dafür sogar Papier-Literatur bzgl. C++ Programmierung gelesen, um überhaupt erstmal ein wenig C++ zu verstehen.

Ja, es ist mühselig, sich erstmal einzuarbeiten. Aber man versteht die Zusammenhänge viel besser, zumal man auch immer die Gegenseite (CCU) betrachten/verstehen/kennen muss. 

Für Detailfragen bzw. Diskussionen sind das Homematic Forum oder das FHEM Forum ein guter Anlaufpunkt.

Und vielleicht bist ja sogar du derjenige, der eine Dokumentation erstellt, wo du dich grad so strukturiert durch die Lib arbeitest :)

_...meine 3 Mark 50_

trilu2000, 2020-11-29:
Vielen Dank Jerome, genau so sehe ich das auch!

> Was schon ein bisserl abschreckend ist wenn derjenige, der AskSinPP laut Source komplett geschrieben hat, sagt "boah, keine Ahnung wie das gehen soll".

Die Asksin ist ein Framework zum Erstellen von Homematic kompatiblen Geräten. Das Framework kümmert sich folglich um die Kommunikation zwischen Zentrale und Peers. Die Kanalparameter sind pro Gerät unterschiedlich und beziehen sich immer auf die spezifische Hardware - d.h. man kann eine Library entwickeln und zur Verfügung stellen ohne sich im Detail mit jedem Gerät und dessen Kanalparameter im Detail beschäftigt zu haben.

Um mal auf die ursprüngliche Frage zu antworten - der original Motion Sensor arbeitet mit einem PIR Element das über einen Verstärker und Comperator an die CPU angebunden ist. Bei so einer Schaltung bekommt man, je nach Intensität der Bewegung, mehrere High/Low Wechsel und genau da greift der Filter mit der Anzahl der High/Low Wechsel ein.
 
In dem Beispiel der Asksin, [MDIR ](https://asksinpp.de/Projekte/jan/HM-SEC-MDIR/), wird ein fertiges PIR Modul genutzt. Dieses gibt bei Bewegung ein HIGH Signal zwischen 2 Sekunden und 1 Stunde, je nach Widerstandswert, aus.
Folglich machen hier Kanalparameter zum Filtern der Anzahl der Impulse überhaut keinen Sinn.

Um noch mal das Thema Dokumentation aufzugreifen - fang doch einfach an, ich bin mir sicher das viele etwas Beitragen werden. Die Kunst ist aber eine Struktur zu finden und diese auch weiter zu verfolgen, die es anderen einfach ermöglicht, sich in das komplexe Thema Homematic einzuarbeiten. 

Aus meiner Sicht ist das Homematic Protokoll sowieso eins der Besten. Es ist schlank und die Geräte können untereinander komplex Verknüpft werden ohne das eine Zentrale eingreifen muss - hier ist es selbst Zigbee und Z-Wave überlegen, einzig in der Netzwerkfähigkeit und Verschlüsselung ist es etwas veraltet...

pa-pa, 2020-11-29:
Niemand zwingt Dich dazu diesen Mist hier zu nutzen. Wenn Du was beitragen willst, kannst Du das gerne tun. Wenn nicht - geh woanders spielen.
Ich habe keine Zeit eine Doku zu schreiben. Habe übrigens nach dem 2. Absatz aufgehört zu lesen. Falls da noch was konstruktives kam, hat es mich leider nicht mehr erreicht.