pa-pa/AskSinPP #180: Weather.h BIDI + sendBroadcast

psi-4ward, 2020-04-03:
Hi,

in https://github.com/pa-pa/AskSinPP/blob/master/Weather.h#L22 wird das `BIDI` Flag gesetzt wobei 
in https://github.com/pa-pa/AskSinPP/blob/master/Weather.h#L44 ein `broadcastEvent()` verschickt wird. Hier gibt also `BIDI` keinen Sinn oder?

Sollten wir nicht die Implementierung von Tom:
https://github.com/TomMajor/SmartHome/blob/master/HB-UNI-Sensor1/HB-UNI-Sensor1.ino#L323
adaptieren und zumindest ca einmal pro Tag an `sendPeerEvent()` schicken?

Somit würden auch eventuelle Config-Daten übertragen werden was aktuell nur per Config-Press möglich ist.

TomMajor, 2020-04-04:
nur zum 2. Punkt von psi:
dieses Feature ist afaik nicht das eq3 Originalverhalten eines Wettersensors sondern ein Enhancement.
Muss man abwägen ob die AskSinPP eher das Original abbilden soll und solche Änderungen nur in abgeleiteten Geräten auftauchen oder nicht..

pa-pa, 2020-04-04:
Hm - broadcastEvent nimmt das BIDI wieder weg (msg.clearAck). Also alles bestens. Die WeatherMessage ist generisch initialisiert und könnte auch direkt an andere Empfänger gesendet werden. Dann wird das BIDI wieder benötigt.
Ich lege keine besonderen Wert darauf, ob AskSin++ exakt wie das Original funktioniert. Hauptsache es ist kompatibel. Können das gern anpassen.

psi-4ward, 2020-04-04:
Rein vom Feature her finde ich diese Lazy-Config schon ziemlich nett. Ich hatte ja ein paar WDS40 mit Quarz aufgebaut und mich damals schon gefragt warum er die Config nicht frisst.
Vllt wäre eine sendPeerEvent ungefähr einmal pro Tag eine gute Alternative bezüglich Stromverbrauch/Sendezeit