pa-pa/AskSinPP #183: AskSinPP-3 oder AskSInPP-4 als ZIP Bibliothek hinzufügen

JurgenO, 2020-05-04:
Hallo,

Das Hinzufügen der ZIP Bibliotheken AskSinPP-3 und AskSinPP-4 schlägt fehl: "Angegebener Ordner/ZIP-Datei enthält keine gültige Bibliothek" 

Ich nutze die Arduino-IDE Version 1.8.12. 

Das Hinzufügen der Bibliotheken AskSinPP-1, AskSinPP-2 sowie AskSin-Master funktioniert.

Kann mir jemand bitte ein Tipp geben woran das liegt oder was ich falsch mache ?

Gruss vom Niederrhein
Juergen

stan23, 2020-05-04:
Wie hast du das ZIP denn runtergeladen?
Über diesen Link?
https://github.com/pa-pa/AskSinPP/archive/V4.zip

JurgenO, 2020-05-04:
Hallo,
danke für die schnelle Antwort. Die Bibliothek über https://github.com/pa-pa/AskSinPP/archive/V4.zip hat die gleiche Fehlermeldung. Ich hatte den Branch V4 ausgewählt und dann die Datei heruntergeladen. Gibt es eine Möglichkeit dass mir die IDE sagt was genau das Problem ist ?

Gruss
Juergen

stan23, 2020-05-04:
Auf den ersten Blick sehen die Archive für V4 und master gleich aus, und alle Branches waren mal der master bevor sie ausgebrancht wurden.

Du könntest ja von der installieren Lib den Inhalt mit dem der V4 austauschen.

JurgenO, 2020-05-04:
Zwischenergebnis: durch "Try and Error" habe ich herausgefunden dass die Arduino IDE das AskSinPP-v4 ZIP Paket akzeptiert, wenn ich das enthaltene Verzeichnis "bootloader" lösche.

Der Master Branch AskSinPP-Master enthält auch ein bootloader Verzeichnis. Jetzt werde ich als nächstes untersuchen, worin sich die Verzeichnisse unterscheiden

JurgenO, 2020-05-17:
Hallo nochmal.
Das AskSinPP-v4 ZIP Paket laesst sich als Bibliothek importieren wenn ich das Verzeichnis 
AskSinPP-4\bootloader\avr\srecord lösche.

ABER: 
Ich kann die AskSinPP-Master Bibliothek ohne Probleme einbinden, obwohl das auch das srecord Verzeichnis beinhaltet.

???

Merkwürdig. Mein Entwicklungsumgebung ist Win10 - das sollte doch kein Problem sein, oder ?

Gruss vom Niederrhein
Jürgen

JurgenO, 2020-05-19:
Hallo, 

das Problem scheint das Laden über das Arduino Menu 
_Sketch > Bibliothek einbinden > .ZIP-Bibliothek hinzufügen_ zu sein.
Wenn ich ich die Bibliothek manuell einbinde wird sie erkannt & kann genutzt werden.

Deshalb denke ich, dass das ein Arduino IDE Problem ist und schliesse das Issue hier.

Gruss vom Niederrhein
Juergen

closed 2020-05-19 by JurgenO