pa-pa/AskSinPP #191: Füllstandssensor mit lokaler Displayanzeige

HMelzer, 2020-06-21:
Hallo,
ich wende mich heute an euch, weil ich mit o.g. Projekt einfach nicht weiter komme:
Ich beabsichtige, eine Füllstandsmessung mittels Ultraschallsensor HC-SR04 zu realisieren, bei gleichzeitiger Temperaturmessung, um die Temperaturabhängigkeit der Schallgeschwindigkeit zu kompensieren, und den Füllstand und die Temperatur lokal auf einem OLED-Display 128x64 anzuzeigen. Dazu verwende ich einen Arduino Pro mini.
Das funktioniert auch hervorragend, wenn ich den folgenden Beispiel-Skech verwende:

`/*
  Fuellstandssensor V.1.0
  Autor: H. Melzer
  Grundlage: Ardudronix
  https://www.ardutronix.de
*/

// Variable deklarieren
int verzoegerung = 5000;
double entfernung = 0.0;
double heightTank=110.0, fuellhoehe;

//Definitionen fuer HC-SR04
#define TRIG_PIN 7
#define ECHO_PIN 6 

//Display-Bibliotheken einbinden
#include <Adafruit_SSD1306.h>
#include <Adafruit_GFX.h>

//Bibliothek für den Temperaturfüler
#include <DallasTemperature.h>

//DS18B20 Sensors verbunden mit Pin 3
OneWire oneWire(3);

//Ultrasonic-Sensor einbinden
#include "src\HCSR04.h"

// Auflösung des Displays setzen
#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels
// Declaration for an SSD1306 display connected to I2C (SDA, SCL pins)
#define OLED_RESET     4 // Reset pin # (or -1 if sharing Arduino reset pin)
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);

#define Dallas

//Initialisierung des Sensor zur Benutzung der Digital-Pins 7 und 6
//UltraSonicDistanceSensor distanceSensor(trigger, echo);
UltraSonicDistanceSensor distanceSensor(TRIG_PIN, ECHO_PIN);
DallasTemperature sens_temp(&oneWire);

void setup()
{
  //Einrichten der Kommunikation mit der Arduino IDE
  Serial.begin (115200);
  Serial.println("Temperatursensor - DS18B20"); 

  //Starten der Kommunikation mit dem Sensor
  sens_temp.begin();
  
//Display mit I2C-Adresse 0x3c initialisieren
    if(!display.begin(/*SSD1306_EXTERNALVCC*/SSD1306_SWITCHCAPVCC, 0x3C)) { // Address 0x3C for 128x64
    Serial.println(F("SSD1306 allocation failed"));
    for(;;);
  }

  display.clearDisplay(); //Verhindert die Anzeige des Adafruit-Logos
  display.display();//Bereitet nächste Anzeige vor
  display.setTextColor(WHITE); //Textfarbe setzen (hier tatsächlich aber nicht weiß
  display.setFont(); //Standard-Text-Font verwenden
}

void loop()
[... 413 lines cut ...]
      Serial.println(F("SSD1306 allocation failed"));
      for(;;);
    }
  display.clearDisplay(); //Verhindert die Anzeige des Adafruit-Logos
  display.display();//Bereitet nächste Anzeige vor
  display.setTextSize(1); //Textgröße setzen
  display.setTextColor(WHITE); //Textfarbe setzen (hier tatsächlich aber nicht weiß
  display.setFont(); //Standard-Text-Font verwenden
  display.setCursor(5, 10); //Begin der Textausgabe setzen

#endif;

}

void loop()
{
  bool worked = hal.runready();
  bool poll   = sdev.pollRadio();
  if (worked == false && poll == false) {
    // deep discharge protection
    // if we drop below critical battery level - switch off all and sleep forever
    // if nothing to do - go sleep
    hal.activity.savePower<SAVEPWR_MODE>(hal);
  }
}`

pa-pa, 2020-06-22:
Ich glaube mich zu erinnern, dass die GFX-Library zur Laufzeit ordentlich Speicher benötigt. Es wird das Bild er im Speicher aufgeb aut und dann zum Display gesendet.
Du könntest mal als Alternative die [u8g2](https://github.com/olikraus/u8g2) Library für die Ausgabe probieren. Diese schreibt direkt in den Display-Speicher.

HMelzer, 2020-06-22:
Nun, das würde den Speicherhunger erklären. Ich habe darüberhinaus schon mehrere unterschiedliche Libraries probiert, aber ohne eine wirkliche Verbesserung. Die u8g2 habe ich auch in meiner IDE, kann jetzt aber nicht sagen, ob ich die schon mal pobiert habe.
Nichtsdestotrotz erklärt es aber nicht das Verhalten, dass die I2C-Adresse unter bestimmten Umständen nicht angesprochen wird und eben auch nicht, dass es keine volle Auflösung am Display gibt, und das eben dann, wenn die AskSinPP-Bibliothek im Spiel ist

HMelzer, 2020-06-22:
Ich habe nun das mit der u8g2-Library versucht. Das hat gut funktioniert, wenn man den richtigen Fonts findet und verwendet. Es geht ohnehin nur mit der U8x8lib.h und eben nicht mit der U8g2lib.h.
Interessant zu wissen wäre allerdings schon, warum es damit geht und nicht mit der Library von Adafruit.
Die u8x8lib bzw. u8g2lib ist übrigens sehr schön vom Autor dokumentiert.

pa-pa, 2020-06-22:
Weil die andere Lib zu viel Speicher braucht. Dann überlagern sich Stack und Heap irgendwann und schon passieren komische Dinge.

HMelzer, 2020-06-24:
Hallo,
ich muss noch einmal den Expertenrat in Anspruch nehmen. Ich habe das Programm und die zugehörige XML-Datei erstellt und in der Raspberrymatik installiert. Der Sensor wird wie erwartet angezeigt und zeigt auch die gemessenen bzw. errechneten Werte:

[image]
Die Einstellungsseite des Gerätes sieht so aus:
[image]

Und nun zum Problem:
Nach dem Anlernen des Gerätes werden nicht die Default-Werte aus der XML-Datei auf der Einstellungsseite übernommen, sondern die Werte, die im Programm stehen und mit denen das Gerät derzeit läuft. Das wäre ja nicht schlimm, wenn man hier einfach ändern könnte. Eine Änderung der Werte für die Sendeversuche und das Sendeintervall wird problemlos an das Gerät weitergeleitet und übernommen, die Werte für die Befestigungs- und die Füllhöhe werden weder von der webui noch vom Gerät übernommen.
Ich dachte, dass ich den Mechanismus verstanden hätte, aber irgendwo muss ja der Haken sein.
Die XML-Datei füge ich hier mal an:
```
<?xml version="1.0" encoding="iso-8859-1"?>
<device version="2" rx_modes="CONFIG,WAKEUP,LAZY_CONFIG" cyclic_timeout="45000">
  <supported_types>
    <type name="HM-Sonic-Level" id="HM-Sonic-Level">
      <parameter index="10.0" size="2.0" const_value="0xF701" />
    </type>
  </supported_types>
  <paramset type="MASTER" id="hm_sonic_level_dev_master">
    <parameter id="TRANSMIT_DEV_TRY_MAX">
      <logical type="integer" min="1.0" max="10.0" default="6.0"/>
      <physical type="integer" interface="config" list="0" index="20" size="1"/>
    </parameter>
    <parameter id="Sendeintervall">
      <logical type="integer" min="10" max="600" default="60" unit="s" />
      <physical type="integer" interface="config" list="0" index="32" size="2" />
    </parameter>
    <parameter id="Befestigungshöhe des Sensors">
     <logical type="integer" min="0" max="4000" default="1800" unit="mm" />
      <physical type="integer" interface="config" list="0" index="34" size="2"/>
    </parameter>
    <parameter id="Max. Füllhöhe">
     <logical type="integer" min="0" max="4000" default="1500" unit="mm" />
      <physical type="integer" interface="config" list="0" index="52" size="2"/>
    </parameter>
  </paramset>
  <channels>
    <channel index="0" type="MAINTENANCE" ui_flags="internal" class="maintenance" count="1">
      <paramset type="MASTER" id="maint_ch_master" />
      <paramset type="VALUES" id="maint_ch_values">
        <parameter id="UNREACH" operations="read,event" ui_flags="service">
          <logical type="boolean" />
          <physical type="integer" interface="internal" value_id="UNREACH" />
        </parameter>
        <parameter id="STICKY_UNREACH" operations="read,write,event" ui_flags="service,sticky">
          <logical type="boolean" />
          <physical type="integer" interface="internal" value_id="STICKY_UNREACH" />
        </parameter>
        <parameter id="CONFIG_PENDING" operations="read,event" ui_flags="service">
          <logical type="boolean" />
          <physical type="integer" interface="internal" value_id="CONFIG_PENDING" />
        </parameter>
        <parameter id="LOWBAT" operations="read,event" ui_flags="service">
          <logical type="boolean"/>
          <physical type="integer" interface="internal" value_id="LOWBAT"/>
        </parameter>
        <parameter id="RSSI_DEVICE" operations="read,event">
          <logical type="integer" />
          <physical type="integer" interface="internal" value_id="RSSI_DEVICE" />
        </parameter>
        <parameter id="RSSI_PEER" operations="read,event">
          <logical type="integer" />
          <physical type="integer" interface="internal" value_id="RSSI_PEER" />
        </parameter>
        <parameter id="DEVICE_IN_BOOTLOADER" operations="read,event" ui_flags="service">
          <logical type="boolean" />
          <physical type="integer" interface="internal" value_id="DEVICE_IN_BOOTLOADER" />
        </parameter>
        <parameter id="UPDATE_PENDING" operations="read,event" ui_flags="service">
[... 27 lines cut ...]
           <physical type="integer" interface="command" value_id="Füllgrad">
             <event frame="FILLING_EVENT"/>
           </physical>
        </parameter>
      </paramset>  
      <paramset type="LINK" id="HM-Sonic-Level_link" />
    </channel>
  </channels>
  <frames>
      <frame id="FILLING_EVENT" direction="from_device" event="true" fixed_channel="1" type="0x53">
         <parameter type="integer" index="9.0" size="2.0" param="TEMPERATURE"/>
         <parameter type="integer" index="11.0" size="2.0" param="Füllhöhe" />
         <parameter type="integer" index="13.0" size="1.0" param="Füllgrad"/>
    </frame>
  </frames>
  <paramset_defs></paramset_defs>
</device>

```

Das Programm hatte ich ja oben bereits gepostet. Wenn es jetzt auch auf die u8g2-Library umgeschrieben ist, ist der Programmteil, welcher die Werte der Liste 0 schreibt und liesst, ja unverändert geblieben.

Eine weitere Frage hätte ich noch:
Bei der Anzeige der Gerätedaten wird die Füllhöhe und der Füllgrad in rechteckige Klammern dargestellt, die Temperatur jedoch nicht. Gibt es hier eine Möglichkeit, das zu ändern und wie wäre diese?
Einen großen Dank im Voraus!

jp112sdl, 2020-06-24:
Warum nimmst du nicht das fertige Ultraschall-Füllstandssensor-Projekt?
https://github.com/jp112sdl/HB-UNI-Sen-LEV-US
und hängst dort Temperatur + Display an?

Für dessen Umsetzung ist es nicht nur mit der XML getan, da muss noch wesentlich mehr angepasst werden.
Siehe
https://github.com/jp112sdl/JP-HB-Devices-addon/blob/master/src/addon/install_hb-uni-sen-lev-us
und
https://github.com/jp112sdl/JP-HB-Devices-addon/blob/master/src/addon/www/config/easymodes/HB-UNI-Sen-LEV-US_master.tcl
... kann sein, dass da noch mehr ist. Weiß ich nicht mehr aus dem Kopf.

Wenn du das Gerät vom Typ her bei `CAPACITIVE_FILLING_LEVEL_SENSOR` belässt, ersparst du dir auch ein wenig Arbeit, was die Darstellung in der WebUI betrifft, weil die internen Programme zum Zusammenstellen der Funktionen korrekt abgearbeitet werden.

Dort ist auch alles "eQ3-konform" was die Bezeichner betrifft.
Umlaute in der XML sind nicht so schön, man arbeitet eigentlich mit Übersetzungstabellen innerhalb der CCU.
Die eckigen Klammern kommen nämlich daher, dass keine Übersetzungen für z.B. `Füllgrad` gefunden wird.

Sind alles nur Anregungen...

HMelzer, 2020-06-24:
Weil es bspw. kein Batteriegerät sein soll, ich ohnehin schon Schwierigkeiten mit dem Speicher des Arduino hatte, nicht sehen kann, wie ich meine Parameter für die Höhen da unterbringe und vielleicht auch noch was anderes.
Trotzdem Danke für die Anregung.

jp112sdl, 2020-06-24:
> Weil es bspw. kein Batteriegerät sein soll

Aber das ist dem Sketch ja egal, ob der Strom aus Batterien oder einem Netzteil kommt ;) 
Lass den LowBat Parameter ruhig drin... dann wird eben niemals LowBat gemeldet.

>  Parameter für die Höhen

Du meinst die "Befestigungshöhe" ? Ist bei mir auch drin und mit `distanceOffset` bezeichnet

> Trotzdem Danke für die Anregung.

Viel Erfolg

HMelzer, 2020-06-24:
Wie gesagt, ich bekomme das im Speicher  nicht unter, da die Anzeige viel davon benötigt.
Allerdings wäre ich schon interessiert daran, zu erfahren, woran meine Lösung krankt und wie das zu korrigieren wäre.

jp112sdl, 2020-06-24:
>  die Werte für die Befestigungs- und die Füllhöhe werden weder von der webui noch vom Gerät übernommen.

Werden sie  denn überhaupt gesendet?
Vielleicht liegt es tatsächlich an den Bezeichnern. Umlaute, Punkt, Leerzeichen

> Bei der Anzeige der Gerätedaten wird die Füllhöhe und der Füllgrad in rechteckige Klammern dargestellt, die Temperatur jedoch nicht. Gibt es hier eine Möglichkeit, das zu ändern und wie wäre diese?

Siehe meinen Erklärungen zu den Übersetzungen weiter oben:
_Die eckigen Klammern kommen nämlich daher, dass keine Übersetzungen für z.B. Füllgrad gefunden wird._

Es gibt 2-3 Stellen (Dateien), an denen Übersetzungen einzutragen sind.
Findest du alles hier https://github.com/jp112sdl/JP-HB-Devices-addon/blob/master/src/addon/install_hb-uni-sen-lev-us

HMelzer, 2020-06-24:
Oh, vielen Dank, die Umlaute waren tatsächlich das Problem.
Das mit den eingeklammerten Bezeichner schaue ich mir aber erst morgen an.
Nochmals vielen Dank, dass es daran gelegen haben könnte, darauf wäre ich wohl nie gekommen.

jp112sdl, 2020-06-28:
Danke auch für die Anregung mit der Temperaturkompensation.
Habe das nun mit in den HB-UNI-Sen-LEV-US aufgenommen und lasse die Temperatur auch gleich noch mit anzeigen (wenn sie ja schon gemessen wird). 
    [image]

closed 2020-09-08 by pa-pa