pa-pa/AskSinPP #192: Support for Sparkfun Pro Micro Board

RainerMinixhofer, 2020-06-21:
Great Library!

I started to implement a Glass Breakage Detection Sensor from Abus by connecting it to an Sparkfun Pro Micro 3.3V Board (VID 1B4F, PID 9204).
When starting to use the AskSinPP library I got stuck with the missing support of this hardware (at least I suppose this is the root cause) in th AskSinPP library.
When compiling some example sketches from the AskSinPP library (e.g. HM-SEC-SD, which I plan to use) within Arduino IDE I get the attached error messages which point into this direction.
When I deliberately change the Board definition from "Sparkfun Pro Micro" to e.g. "Arduino Mini" the compile runs through without any error message.
I noticed that the same errors occur with the Board setting to "Arduino Micro" btw.

I'm rather new to Arduino programming, thus help would be appreciated here.

Best regards,

Rainer
[error_log.txt](https://github.com/pa-pa/AskSinPP/files/4810081/error_log.txt)

jp112sdl, 2020-06-21:
Hi,
the AskSinPP library does not support the 32U4 microcontroller.
You need one of the controllers mentioned in the README:
https://github.com/pa-pa/AskSinPP#asksin--
- ATMega328
- ATMega32 (Standard Pinout)
- ATMega644 (Bobuino Pinout is highly suggested)
- ATMega1284 (No OTA Bootloader support, yet)
- STM32F1

RainerMinixhofer, 2020-06-21:
Hi Jerome,

I noticed the missing support for the 32U4 microcontroller after I have wired up the hardware already. Any plans to add this support in future?

Rainer

TomMajor, 2020-06-21:
Der 32U4 wurde u.a. auch hier schon mal diskutiert:
https://homematic-forum.de/forum/viewtopic.php?f=76&t=53864&p=536675&hilit=32u4#p536675

RainerMinixhofer, 2020-06-22:
Danke für den Link Tom,

 aufgrund der Kompilerfehler sieht es nach ein paar (überschaubaren) Problemen in "AlarmClock.h" "Activity.h" und "Lowpower.h" aufgrund der klarerweise verschiedenen Registerdefinitionen aus. Das Errorlog schlägt ja sogar schon Register vor die beim ATmega32U4 vorhanden sind. 
Mir ist nur nicht klar ob es hier nicht Hardwareinkompatibiliäten zwischen den ATmega32U4 und z.B. dem ATmega328B gibt oder ob ich in den entsprechenden Headerfiles der AskSinPP einfach die Register über #if includes für den ATmega32U4 anpassen kann?

pa-pa, 2020-06-22:
Probier doch erst mal - wenn es läuft, können wir immer noch sehen, wie wir das am Besten integrieren.

TomMajor, 2020-06-22:
Bei Änderungen/Anpassungen der Regs würde ich grundsätzlich prüfen ob dabei die Belegung der U4 Regs denen des 328P entsprechen.
Außerdem gibt es keinen Timer2 beim U4, dafür aber Timer 3 und 4.

RainerMinixhofer, 2020-06-22:
Habe die Unterschiede in der Timerbelegung schon bemerkt. Es tauchen in der Fehlermeldung aber auch noch Fehler mit dem UART Registern auf. Aber das scheint es auch schon gewesen sein. Ich schaue mir an ob ich einen Pull Request in den nächsten Tagen zusammenkriege, nachdem ich die Library auf dem ATmega32U4 zum Laufen bekomme...