pa-pa/AskSinPP #203: Cannot assign HM Key

oskarn97, 2020-09-17:
Hey there,
I am trying to recreate a HM-RC-KEY4-3. Unfortunately, when issuing the assignHmKey command in with my CUL, my module responds with NACK. How do I activate the storage of 

`const struct DeviceInfo PROGMEM devinfo = {
  DEVICE_ID,     // Device ID
  SERIAL,           // Device Serial
  {0x00, 0xD6},           // Device Model
  0x11,                   // Firmware Version
  as::DeviceType::Remote, // Device Type
  {0x00, 0x00}            // Info Bytes
};
typedef AvrSPI<10, 11, 12, 13> SPIType;
typedef Radio<SPIType, 2> RadioType;
typedef AskSin<StatusLed<7>, BatterySensorExt<A0, 0xFF, 1>, RadioType> HalType;

class Hal : public HalType {
    AlarmClock btncounter;
  public:
    void init (const HMID& id) {
      HalType::init(id);
      battery.init(3, btncounter);
      battery.low(36);
      battery.critical(34);
    }

    void sendPeer () {
      --btncounter;
    }

    bool runready () {
      return HalType::runready() || btncounter.runready();
    }

};

typedef RemoteChannel<Hal, PEERS_PER_CHANNEL, List0> ChannelType;
typedef MultiChannelDevice<Hal, ChannelType, 4> RemoteType;

Hal hal;
RemoteType sdev(devinfo, 0x20);
ConfigButton<RemoteType> cfgBtn(sdev);
`

Thanks for your help.

jp112sdl, 2020-09-17:
You should also check the `BatterySensorExt` template parameters.
reference voltage 1 seems to be not correct?

And did you wake up the HM-RC-KEY4-3 by pressing the CONFIG_BUTTON before trying the assignHmKey command?

jp112sdl, 2020-09-17:
...and one more idea: di you run the FreqTest sketch to get the correct cc1101 frequency settings?

oskarn97, 2020-09-17:
That is just the divider ration, since I have a battery lower than Vcc, I do not need a voltage divider. Communication in general works fine, only assigning a new HM Key does not work.

jp112sdl, 2020-09-17:
> That is just the divider ration

https://github.com/pa-pa/AskSinPP/blob/master/BatterySensor.h#L247-L248
```
template <uint8_t SENSPIN,uint8_t ACTIVATIONPIN,uint16_t VCC=3300>
class BatterySensorExt : public BatterySensor {
```
The 3rd parameter is the VCC reference value

> only assigning a new HM Key does not work.

but you did compile the sketch with `#define USE_AES` ?
https://github.com/pa-pa/AskSinPP#enable-aes-support

pa-pa, 2020-09-17:
1 minute before me :-)
You need AES support compiled in.

oskarn97, 2020-09-17:
Yes, I did and I also provided a default AES Key. What I want to do now is transmit a custom AES Key without compiling it in. Is that possible?

oskarn97, 2020-09-18:
I traced it back to this line: https://github.com/pa-pa/AskSinPP/blob/ecdefd730f62b5616c92b65abbf14eac9eca2ec0/Sign.h#L120
This check fails. What does that mean?

jp112sdl, 2020-09-18:
Maybe have to compile with the correct eQ-3 default AES key (can be found with Google).
`#define HM_DEF_KEY 0x01,0x02,0x03,0x04,0x05,0x06,0x07,0x08,0x09,0x0a,0x0b,0x0c,0x0d,0x0e,0x0f,0x10` is just fake data.
Using this fake aes key causes "Signature FAILED" on my site.
[image]

But only for the case, there is no custom key set while teaching in.

If you have already a custom key set while teaching, you can use the fake key above.

[image]

_All tests were done with RaspberryMatic._

oskarn97, 2020-09-18:
Hey Jerome,

thanks for your reply.
When using an example Sketch like HM-SEC-WDS.ino, everything works.
With remote channel applications like HM-RC-4.ino it does not.
I've provided the correct AES Key.

To reproduce this issue, please try the HM-RC-4.ino sketch.
[image]

jp112sdl, 2020-09-18:
> please try the HM-RC-4.ino sketch.

I used the HM-RC-4 sketch for the tests and screenshots above

oskarn97, 2020-09-18:
Providing the correct default key seemed to have worked for me... Thanks a lot!!

closed 2020-09-18 by oskarn97