pa-pa/AskSinPP #204: Recreating HM-RC-Key4-2

oskarn97, 2020-09-20:
Hey there,

I am trying to recreate HM-RC-Key4-2 and peer it directly to KeyMatic without the help of a VCCU.
I've deployed the default eq3 AES Key and enabled AES support inside the sketch.
Pairing it to FHEM and then peering it to KeyMatic works fine, but the direct peering does not.
Once the pairing mode of an additional remote in KeyMatic is enabled, I start the pairing using the config button at my PCB (led blinks). Pressing any button of the recreated remote also does nothing.
I would be very thankful for any advice.

Thanks a lot!

```
#define SERIAL "ABCD"
#define DEVICE_ID {0x10, 0x08, 0x05}

#define USE_AES
#define HM_DEF_KEY 0xA4,0xE3... //default eq3 key
#define HM_DEF_KEY_INDEX 0

const struct DeviceInfo PROGMEM devinfo = {
  DEVICE_ID,     // Device ID
  SERIAL,           // Device Serial
  {0x00, 0xA6},           // HM-RC-Key4-2 
  0x11,                   // Firmware Version
  as::DeviceType::Remote, // Device Type
  {0x00, 0x00}            // Info Bytes
};
typedef AvrSPI<10, 11, 12, 13> SPIType;
typedef Radio<SPIType, 2> RadioType;
typedef AskSin<StatusLed<LED_PIN>, BatterySensorExt<A0, 0xFF>, RadioType> HalType;

typedef RemoteChannel<Hal, PEERS_PER_CHANNEL, List0> ChannelType;
typedef MultiChannelDevice<Hal, ChannelType, 4, List0> RemoteType;

Hal hal;
RemoteType sdev(devinfo, 0x20);
ConfigButton<RemoteType> cfgBtn(sdev);

void setup()
{
  sdev.init(hal);
  buttonISR(cfgBtn, CONFIG_BUTTON_PIN);
  cfgBtn.init(CONFIG_BUTTON_PIN);
  sdev.initDone();
}

```

jp112sdl, 2020-09-20:
You know that the SERIAL has to be a string with a fixed length of 10 characters?

oskarn97, 2020-09-20:
Yes, that was just a placeholder

oskarn97, 2020-09-20:
I've just read that it is not possible to directly peer devices without the help of VCCU. Is that information still up to date?

jp112sdl, 2020-09-20:
If the device is paired with a ccu, you cannot create peering between devices directly.
That is still up to date.
[image]

oskarn97, 2020-09-20:
That is not what I meant. I read in a forum that AskSinPP cannot peer devices directly without the help of CCU.
The Keymatic is only paired to a master remote.

jp112sdl, 2020-09-20:
>  that AskSinPP cannot peer devices directly 

Ah okay, sorry, then unfortunately I can't help. I never tried to peer devices directly without a central

oskarn97, 2020-09-20:
Thanks anyway, appreciate it!

pa-pa, 2020-09-20:
I have no idea how peering without central should work. I have nothing implemented into this direction.

oskarn97, 2020-09-21:
Would it help if I recorded the communication between an original HM-RC-Key4-2 and a Keymatic while pairing with each other?