pa-pa/AskSinPP #209: Sind direkte Verknüpfungen möglich?

akula3, 2020-10-09:
Hallo,

ich habe den HB-UNI-Sen-RFID-RC von Jérôme nachgebaut (vielen Dank an Ihn für dieses geniale Projekt!).
AES im Scetch ist aktiviert, anlernen an die CCU2 ohne Fehler geklappt.
Die RFID Tags lassen sich registrieren und eine anmeldung von einem RFID Tag wird auch in der CCU WEB UI angezeigt.
Ein angelegtes Programm mit HB-UNI-Sen-RFID-RC und HM-Sec-Key funktioniert auch.
Eine direkte Verknüpfung lässt sich zwischen den beiden anlege, funktioniert aber nicht.
Der serielle Monitor am Arduino meldet auch keine Fehler.
Das alles mit eingestellten Übertragungsmodus "Standard" bei HB-UNI-Sen-RFID-RC.
Sobald ich den Übertragungsmodus beim HB-UNI-Sen-RFID-RC auf "Gesichert" ändere, Konfigurationsdaten für HB-UNI-Sen-RFID-RC stehen zur Übertragung an, und die Meldung erlischt nich weder beim annähern des RFID Tag noch durch drücken der Anlerntaste beim HB-UNI-Sen-RFID-RC.
Es gibt auch Fehlermeldungen im seriellen Monitor "CRC Failed" und "Signature FAILED".

Den AES Key habe ich im Scetch wie folgt angelegt:
Den System-Sicherheitsschlüssel aus CCU2 einfach StrToHex umgewandelt (als Beispiel 2Enten->32456E74656E) und im Format `#define HM_DEF_KEY 0x01,0x02,0x03,0x04,0x05,0x06,0x07,0x08,0x09,0x0A,0x0B,0x0C,0x0D,0x0E,0x0F,0x10` eingetragen.
Den Key Index habe ich aus \usr\local\config\keys genommen:
```
Current Index = 3
Key 1 = xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Key 2 = xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Key 3 = xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Last Index = 2
```
und als `#define HM_DEF_KEY_INDEX 6` eingetragen, da es in der Info steht, man solle den aktuellen Index mit 2 multiplizieren.
Key Verifikation mit `crypttool -v -t 3 -k 'mein_Schlüssel'` war OK.

Was mache ich falsch?

jp112sdl, 2020-10-09:
> Den System-Sicherheitsschlüssel aus CCU2 einfach StrToHex umgewandelt

Das bringt dir nichts. 
Du musst du den Key nehmen wie er in /etc/config/keys steht.
https://homematic-forum.de/forum/viewtopic.php?t=44178#p441085

Alternativ kannst du auch den Default AES Key ergoogeln und mit HM_DEF_KEY_INDEX 0 nutzen.
Es wird dann beim Anlernen des Geräts automatisch dein persönlicher Schlüssel ins Gerät geschrieben

akula3, 2020-10-09:
Vielen Dank, Jérôme! Du hast mir schon wieder geholfen :-)
Das hat geklappt:
> Alternativ kannst du auch den Default AES Key ergoogeln und mit HM_DEF_KEY_INDEX 0 nutzen.

Einmal zum Verstehen:
> Du musst du den Key nehmen wie er in /etc/config/keys steht.

In der Datei stehen nur die Prüfsummen von den realen Keys, soweit ich weiss.
Willst du damit sagen, daß die Prüfsumme anstatt dem realen Key in den Sketch kommt?

Und was ist mit:
> The index is the key number multiplied by 2.

Ist es noch richtig?

jp112sdl, 2020-10-09:
> In der Datei stehen nur die Prüfsummen von den realen Keys, soweit ich weiss.

Da stehen die Keys. Alles andere wäre mir neu. 

> Ist es noch richtig?

Glaube ja. Habe ich aber auch bisher nur ein Mal vor Jahren genutzt. 
Nehme wenn dann den Default Key mit Index 0... den Rest macht die CCU bzw. AskSin++ ja dann von selbst

akula3, 2020-10-09:
> Da stehen die MD5 Sums der Keys.

Zumindest hier hatte ich recht :-)
Das ist mir aber irgendwie untergegangen:
> Und das muss auch in den Sketch rein

Aber egal, Du hast absolut recht:
> Nehme wenn dann den Default Key mit Index 0... den Rest macht die CCU bzw. AskSin++ ja dann von selbst

So ist es viel einfacher und man braucht keine Gedanken um eigenen Key und dessen Index mehr zu machen.

Nochmals vielen Dank!!!

closed 2020-10-09 by akula3