pa-pa/AskSinPP #210: how is it possible to use the AskSinPP libary with the original trx-868 RF module

v0r73x, 2020-10-16:
Hi,

Is it possible to use the EQ3 RF modules (trx868-TFK-TI or trx868-SL) instead of the cc1101 rf module?
If yes, how i can do that? 

thanks for your feedback,

v0r73x

jp112sdl, 2020-10-16:
Hi,
yes it is possible.
wiring is explained here: https://wiki.fhem.de/wiki/HomeMatic_Asksin_Library

v0r73x, 2020-10-16:
i cabeled it already and tested it but i see only the CC init and that is all ;(

v0r73x, 2020-10-16:
[image]

and cabeling as described on the page you given. 
I tried with the freq test first to check the module before starting to attach sensors etc.

jp112sdl, 2020-10-16:
Can you attach a picture of your cabling?

v0r73x, 2020-10-16:
[image]

v0r73x, 2020-10-16:
[image]

v0r73x, 2020-10-16:
[image]

v0r73x, 2020-10-16:
[image]

v0r73x, 2020-10-16:
[image]

v0r73x, 2020-10-16:
von links nach rechts

trx868 PIN1 = rot an GPIO 11
trx868 PIN2 = sw  an GPIO 13
trx868 PIN3 = lila an GPIO 12
trx868 PIN4 = frei 
trx868 PIN5 = blau an GPIO 2
trx868 PIN6 = grün an GPIO 10
trx868 PIN7 = gelb an GPIO GND
trx868 PIN8 = braun an GPIO VCC

jp112sdl, 2020-10-16:
Hmm, ja alles richtig.

Was passiert wenn du die SPI-Pins (11 und 12) trennst?
Startet der Pro Mini dann weiter? Es sollten dann zumindest jede Menge Read-Error kommen

v0r73x, 2020-10-16:
ja er startet und auch die errors kommen wenn ich PIN 11 & 12 abziehe

v0r73x, 2020-10-16:
[image]

jp112sdl, 2020-10-16:
Hat vielleicht doch das Funkmodul einen weg?
Ich vermute mal, dass es bei angeschlossenem Funkmodul an [`waitMiso ()`](https://github.com/pa-pa/AskSinPP/blob/0f3dad518f8bed6f40353473111e06c0f73f52b2/Radio.h#L181) hängt.

v0r73x, 2020-10-19:
Hi Jerome, und alle anderen die mitlesen ;) 

ich habe es jetzt mit einem komplett anderen TRX868 probiert - nun ist der output gleich von begin Fehlerhaft nachdem ich das FreqTest sketch geflashed habe ... kabel sind aber richtig angeschlossen nr PIN 4 vom trx868 habe ich freigelassen 

[image]

v0r73x, 2020-10-19:
Muss ich vielleicht noch etwas im code anpassen damit das trx868 funktioniert?

v0r73x, 2020-10-19:
Ich habe auch die Kabel gekürzt beim anderen trx868 Modul
[image]

jp112sdl, 2020-10-19:
> Muss ich vielleicht noch etwas im code anpassen damit das trx868 funktioniert?

Nein, ich nutze es auch seit Jahren ohne irgendwelche Mods

Ähm... ist das ein TRX868**-TI** oder **-SL** ?

Das **-SL** funktioniert **nicht** mit der AskSinPP. Es besitzt keinen CC1101 Chip!

jp112sdl, 2020-10-19:
Ah, ich sehe etwas unscharf auf dem Foto weiter oben... das ist -SL
Dann brauchst du es nicht weiter versuchen

v0r73x, 2020-10-19:
Ach mist - ok ich habe schon an meinen löt kenntnissen gezweifelt ;) So richtig geht das aber aus der Beschreibung auf https://wiki.fhem.de/wiki/HomeMatic_Asksin_Library nicht hervor, da die "nur" vom breiten und vom schmalen modul sprechen...

v0r73x, 2020-10-19:
Ich habe noch ein Modul gefunden denkst Du das könnte gehen?
[image]

jp112sdl, 2020-10-19:
Dort steht `-TI`
Die Frage solltest du dir somit beantworten können.

v0r73x, 2020-10-19:
ja aber das TFK hat mich jetzt irritiert - ok ich versuche es jetzt damit nochmal

closed 2020-10-19 by v0r73x

jp112sdl, 2020-10-19:
Nur noch mal der Vollständigkeit halber: 
Es gibt beide Modulausführungen (breit und schmal) mit Funkchip von Texas Instruments (`-TI`) und SiliconLabs (`-SL`).
Das breite Modul geht daher prinzipiell auch, sofern es sich um ein `-TI` handelt.

[image]

@stan23 Hattest du nicht auch mal irgendwo einen Beitrag zu den beiden Modultypen geschrieben?

TomMajor, 2020-10-19:
Ich kann die Info im FHEM Wiki ergänzen.
Gibt es eigentlich HM Geräte die immer die -TI bzw. -SL haben oder kann man das vorher nicht sagen bevor man es nicht geöffnet/ausgelötet hat?

stan23, 2020-10-19:
> 
> @stan23 Hattest du nicht auch mal irgendwo einen Beitrag zu den beiden Modultypen geschrieben?

Hier steht was kurzes:
https://homematic-forum.de/forum/viewtopic.php?f=76&t=59292&p=587278&hilit=SiLabs#p587276

Angeschaut habe ich das im Zuge der WoR-Analyze, kann mich aber nicht erinnern ob ich das noch detaillierter dokumentiert habe.

v0r73x, 2020-10-19:
Hi TomMajor, 
ich habe die Beiden TRX868-SL auf dem Aktor: HM-LC-Sw4-Ba-PCB gefunden. 
Beide HM-LC-Sw4-Ba-PCB wurden als defekt verkauft, aber ich dachte ich kann den Funkmodulen nochmal ein neues Leben einhauchen für einen Homebrew Aktor - Schade ;(

jp112sdl, 2020-10-19:
> Gibt es eigentlich HM Geräte die immer die -TI bzw. -SL haben oder kann man das vorher nicht sagen bevor man es nicht geöffnet/ausgelötet hat?

Innerhalb eines Gerätetyps ist das Modul identisch. Der HM-LC-Sw4-Ba-PCB hat dann also immer ein `-SL` verbaut.

In manchen Bauanleitungen steht es detailliert drin, welches Modul verbaut ist, in anderen wiederum nur allgemein "TRX868".

Hier mal was ich auf die Schnelle zusammentragen konnte:
|Gerät|Modul|
|------|------|
|HM-SenDB-PCB|TRX868-SL|
|HM-ES-TX-WM|TRX868-SL|
|HM-Dis-WM55|TRX868-TFK-SL|
|HM-PB-2-FM|TRX868-TFK-SL|
|HM-TC-IT-WM-W-EU|TRX868-TFK-SL|
|HM-CC-RT-DN|TRX868-TFK-SL|
|HM-Sec-SCo|TRX868-TFK-SL|
|HM-RC-2-PBU-FM|TRX868-TFK-SL|
|HM-Sen-MDIR-O|TRX868-TFK-TI|
|HM-SCI-3-FM|TRX868-TFK-TI|
|HM-LC-Sw1-BA-PCB|TRX868-TI|
|HM-OU-CFM-TW|TRX1-TIF|
|HM-LC-Sw1-PLl-DN-R1|TRX1-TIF|
|HM-Dis-EP-WM55|TRX2-TIF|
|HM-LC-Sw2PBU-FM|TRX2-TIF|
|HM-LC-RGBW-WM|TRX3-TIF|

Bei den Bezeichnungen TRX _n_-TIF weiß ich nicht, was genau an Hardware drin steckt.
Ich gehe von einem CC1101 aus... aber ich glaub die 3. Gen. (TRX3) wird auch bei den HmIP-Geräten verwendet und beherbergt ein CC1310? Nur eine vage Vermutung.

TomMajor, 2020-10-19:
Sehr interessant, danke euch Jerome und v0r73x für die Infos zu den Funkchips in HM Geräten.

Ich habe einen Hinweis im wiki ergänzt, neuer Abschnitt **Achtung, verschiedene Funkchips**
https://wiki.fhem.de/wiki/HomeMatic_Asksin_Library

jp112sdl, 2020-10-19:
> Ich habe einen Hinweis im wiki ergänzt

Sehr gut!
_Eigentlich komisch, dass das Thema bisher nie aufkam. Das `-SL` ist ja doch sehr verbreitet_

v0r73x, 2020-10-19:
scheinbar basteln viele nur nach und probieren nicht kaputte aktoren zu upcyclen wie ich ;) 

@TomMajor besten dank für das Einpflegen ins wiki :) 

OFFTOPIC ich werde jetzt tiefer in die HomeBrew Thematik einsteigen - Ich würde gern den HB-UNI-RGB-LED-CTRL 
mit einem ds18b20 temp fühler austatten wollen da der im wohnzimmer hängt - @jp & all others: ist das generell möglich ?

jp112sdl, 2020-10-19:
Mach für allgemeine Diskussionen bitte einen neuen Thread im Homematic Forum auf
https://homematic-forum.de/forum/viewforum.php?f=76

Um es vorweg zu nehmen - ja es ist technisch möglich. Aber der Aufwand, es in die CCU zu integrieren, lohnt nicht. +Nimm am besten 2 einzelne Geräte

TomMajor, 2020-10-19:
Gerne.
Und noch ein Hinweis für @v0r73x 
Du solltest die github Namen exakt schreiben mit @
Momentan referenzierst du Julien Pellet in Montreal.