pa-pa/AskSinPP #220: AskSinPP/examples/stm32/HM-LC-DW-WM/HM-LC-DW-WM.ino

elmarfaber, 2020-12-25:
Hallo,

beim compilieren unter PlatformIO tritt folgender Fehler auf:
Die Datei "Quelle" kann nicht geöffnet werden: "OneWireSTM.h".

Unter PlatformIO finde ich diese Library nicht. Unter rogerclarkmelbourne Arduino_STM32 finde ich diese Library
aber binde ich diese Library ein bekomme ich sehr viele Fehler angezeigt. 

Wie löse ich das Problem?

pa-pa, 2020-12-25:
Mit PlatformIO kenne ich mich nicht aus.
Was sind das denn für Fehler ?

jp112sdl, 2020-12-25:
Die `OneWireSTM.h` ist ja auch eigentlich direkt schon beim Boardsupport [stm32duino](https://github.com/stm32duino/Arduino_Core_STM32) mit drin.
`./packages/stm32duino/hardware/STM32F1/2020.8.9/libraries/OneWireSTM/src/OneWireSTM.h`

Das ist keine extra Lib, die eingebunden werden muss

elmarfaber, 2020-12-26:
Dann mache ich scheinbar etwas grundsätzliches falsch. Ich verwende die Platform ST STM32 und das Board genericSTM32F103C8.
Dabei fehlt die o.g. Library. Ich finde kein stm32duino Bordsupport bei meiner Installation. Wie gehe ich denn hierbei vor? Ich würde gerne weiterhin PlatformIO verwenden - ist einfach viel besser wie die Arduino Umgebung. Die Fehler die angezeigt werden sind sehr viele - so sehzen die ersten aus:
"message": "'PWM16' was not declared in this scope",
"message": "expected unqualified-id before '>' token",
"message": "'class as::InternalEncoder<as::DimmerDevice<as::AskSin<as::StatusLed<32>, as::NoBattery, as::Radio<as::LibSPI<42>, 46> >, as::DimmerChannel<as::AskSin<as::StatusLed<32>, as::NoBattery, as::Radio<as::LibSPI<42>, 46> >, 6>, 6, 3> >' has no member named 'peer'",
u.s.w.

elmarfaber, 2020-12-27:
So wenn ich in der platform.ini folgendes eintrage wird das Projekt compiliert:

[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
board_build.core = maple
monitor_speed = 57600
lib_deps = 
	AskSinPP