pa-pa/AskSinPP #221: Arduino Uno Errors

ednarb29, 2020-12-31:
Hi, I have tried to realize the DHT22 project on my Arduino Uno, but I get several error messages:

`AskSin++ V4.1.7 (Dec 31 2020 14:16:05)
Address Space: 32 - 73
CC init1
Error at 00 expected: 2E read: 00
Error at 02 expected: 06 read: 00
Error at 03 expected: 0D read: 00
Error at 04 expected: E9 read: 00
Error at 05 expected: CA read: 00
Error at 07 expected: 0C read: 00
Error at 0B expected: 06 read: 00
Error at 0D expected: 21 read: 00
Error at 0E expected: 65 read: 00
Error at 0F expected: 6A read: 00
Error at 10 expected: C8 read: 00
Error at 11 expected: 93 read: 00
Error at 12 expected: 03 read: 00
Error at 15 expected: 34 read: 00
Error at 17 expected: 03 read: 00
Error at 18 expected: 18 read: 00
Error at 19 expected: 16 read: 00
Error at 1B expected: 43 read: 00
Error at 1E expected: 2F read: 00
Error at 1F expected: 65 read: 00
Error at 20 expected: 78 read: 00
Error at 23 expected: E9 read: 00
Error at 24 expected: 2A read: 00
Error at 25 expected: 1F read: 00
Error at 26 expected: 11 read: 00
Error at 3E expected: 03 read: 00
CC Version: 00
Error at 3E expected: C0 read: 00
 - ready
iVcc: 4918
Measure...
DHT22 measurement ok.
T/H = 244/41`

Any idea, what could cause this problem?

jp112sdl, 2020-12-31:
Init read with all 0x00 _could_ be a shortcut between MISO and GND.

But don't forget: 
The UNO works with 5V pin levels. The CC1101 works only with 3.3V (max. 3.9V)
It's also possible, that your CC1101 is faulty now...

ednarb29, 2021-01-01:
Sorry for the beginner questions, but you mean all pins, right?

I’ve connected the power pin of the cc1101 to the arduino 3.3v pin, but all other pins are connected as normal. So they might be fried now?

Is there a way to realize this with an uno and the 5v pins actually?

jp112sdl, 2021-01-01:
> but you mean all pins, right

Yes

> Is there a way to realize this with an uno and the 5v pins actually?

You have to use a level shifter for all I/O pins
i.e. https://www.amazon.de/4-Kanal-Converter-Arduino-Raspberry-Mikrocontroller/dp/B01MTUKUUH

TomMajor, 2021-01-01:
Here is an example where I did the level shifting for a 5V Nano:
https://github.com/TomMajor/SmartHome/tree/master/PCB/AskSin-Analyzer-XS

ednarb29, 2021-01-01:
This looks quite complex. My goal is to read an ultrasonic sensor hc sr04 which is 5v I believe.

Do you have an idea how I can combine the 3.3v with the 5v system best? Either 3.3v arduino, 3.3v cc1101 and 5v ultrasonic sensor or 5v arduino, 5v ultrasonic sensor and 3.3v cc1101?

jp112sdl, 2021-01-01:
There is no need to use 5V

Have a look at
https://github.com/jp112sdl/HB-UNI-Sen-LEV-US

P.S.: The US-100 (the on I am currently using) is more reliable than the HC-SR-04 and works also with 3.3V.

ednarb29, 2021-01-01:
Thanks, I’ll give it a try. Do you know where I can get this in Germany a bit faster delivered than the end of the month with Ali or Amazon?

jp112sdl, 2021-01-02:
Have a look at ebay:
https://www.ebay.de/itm/US-100-Ultraschall-Modul-Entfernungsmesser-Sensor-Arduino-Raspberry-Pi-Robotik/174549840779

closed 2021-01-17 by ednarb29