pa-pa/AskSinPP #224: Use CC1310

brandmeister91, 2021-01-18:
I would like to use this library with an CC1310.
Have you already thought about adding support for this controller?
(I know this would be a big thing with a lot of differences)
Could you give a hint, which modules would have to be changed?

TomMajor, 2021-01-18:
> Could you give a hint, which modules would have to be changed?

Radio.h

trilu2000, 2021-01-19:
Asksin is quite hardware independent, as most of the abstraction layer is managed by Arduino.
Nevertheless, there are some hardware references in various modules (Storage.h, AlarmClock.h, BatterySensor.h, Radio.h).

To implement a CPU with new technology, the CC1310 is based on Cortex-M3, it also needs the corresponding Arduino support. As far as I have seen, the CC1310 is currently "only" supported by Energia which is an Arduino fork but has his own development route since several years. You might will have the challange that parts of the asksin framework will not work as expected.

Once all these hurdles have been overcome, the radio still needs to be adapted. In the CC1310, the radio has its own CPU that  needs to be programmed accordingly.

From my point of view, it would be easier to implement an STM32L151 with CC1101.

brandmeister91, 2021-01-19:
Thank you for your answers so far.
I thoght about AskinPP as a simple library, that I could use in other IDEs as well, 
but yeah should watch out for special keywords and compiler specific things,....
Maybe I will look if there is a possibility to use the cc1310 with arduino IDE.
As far as I can see now the Radio things are deeper than only in the Radio.h, since it dosent completely abstracts it?
(For example registers are set from other moduls and not only do a "set ....")

jp112sdl, 2021-01-19:
> Maybe I will look if there is a possibility to use the cc1310 with arduino IDE.

I also already did. 

> As far as I have seen, the CC1310 is currently "only" supported by Energia which is an Arduino fork but has his own development route since several years.

And I also did not find anything that serves a full (or almost complete) integration of the ARM Cortex-M3.

At the moment, I'm trying to do first steps with the TI Code Composer Studio (_TI has a cool cloud-integrated IDE_), but only to learn something about the MSP430.

trilu2000, 2021-01-19:
> And I also did not find anything that serves a full (or almost complete) integration of the ARM Cortex-M3.
stm32duino with the STM32 core is claiming it - they are supporting the STM32L151/152 which is based on Cortex-M3 as well.

I have looked into it over the last couple of days because it would by a good base for further asksin developments. 
Low cost CPU, enough RAM, Flash and very import 8KB EEPROM (CC1310 has no eeprom), several Timers and 16bit PWM and much more.
Unfortunately i didn't have the skills to implement it into the asksin library at the moment as i am struggling on the adoptions in the different modules mentioned above. The only thing i got working so far is the eeprom :-)

Timer, SPI interface, etc is still a challange...

trilu2000, 2021-02-07:
STM32L151 based on arm cortex M3 is implemented now.
It is based on the STM32duino core v1.9.0 for Arduino.
Works quit well so far, Lowpower needs some further optimization.
As next we need a common development HW and an own board definition in the STM32duino core.