pa-pa/AskSinPP #230: Vorgehensweise bei Änderungen

jp112sdl, 2021-02-07:
Hallo zusammen
@pa-pa @trilu2000 

Da nun hier mehrere Leute direkten Schreibzugriff haben, hätte ich vorgeschlagen, dass Änderungen vielleicht erst in einem PR verfasst (oder einem extra Branch) werden und dieser dann (gern auch selbstständig) erst nach *erfolgreichem* Travis-Lauf (der ja automatisch startet) merged wird.

Diese Änderung führt aktuell dazu, dass einige Projekte nicht mehr kompilieren:
https://github.com/pa-pa/AskSinPP/blame/0b8ffaa5f6ce9c40c66356aae7fccbd89f1d1291/Peer.h#L21

Das würde dann vielleicht eher auffallen und somit wäre dann der master-Branch erstmal nicht "kaputt".

pa-pa, 2021-02-07:
Bin im Prinzip auch dafür - wollt @trilu2000 das Leben etwas einfacher machen.
Ich Push ja auch immer direkt.

trilu2000, 2021-02-07:
Ihr müsst mir bitte erklären wie ich mein Zeug hochladen soll. 
Bei mir kompiliert das ja, sonst würde ich es nicht hochladen.
Seit dem letzten Problem mit den überschneidenden Variablennamen kompiliere ich 2 oder 3 verschiedene Sketches auf Basis AVR vorher.

jp112sdl, 2021-02-07:
> Bei mir kompiliert das ja, sonst würde ich es nicht hochladen.

Hmm... aktuell gehen laut Travis 15 Projekte nicht mehr

> Ich Push ja auch immer direkt.

Ist ja auch dein Repo 😎 

Ich hab ja anfangs auch direkt gepusht, aber es gab halt auch oft noch Nacharbeiten, nach Austausch und Diskussionen.
Diese Änderungen dann alle in einem finalen PR zusammengefasst machen die Gesamtstruktur übersichtlicher und nachvollziehbarer.

trilu2000, 2021-02-07:
und was mache ich jetzt?
kann ich den Commit zurück nehmen?

Um einen PR zu machen muss ich wieder einen eigenen fork machen, gegen den mache ich meine Änderungen und dann per PR in den master pushen?

jp112sdl, 2021-02-07:
@pa-pa hat einen fix gepusht, aktuell sieht es gut aus. 
Travis baut aber momemtan noch

jp112sdl, 2021-02-07:
@trilu2000 Du kannst auch hier direkt in einen neuen Branch pushen (nennst ihn _trilu_dev_ oder so) und diesen Branch kannst du dann in den master mergen.

stan23, 2021-02-07:
> 
> Um einen PR zu machen muss ich wieder einen eigenen fork machen, gegen den mache ich meine Änderungen und dann per PR in den master pushen?

Github Desktop bietet das inzwischen an. Man kann das Haupt-Repo für Pulls nehmen, und in einen eigenen Fork pushen und dann einen PR stellen. Oder man macht das per Hand über die Kommandozeile.

Gerade wenn mehrere an einem Projekt arbeiten, sind eigene Branches (auf denen dann die Tests laufen und vor dem Merge ein Review gemacht wird) state of the art, oder?

jp112sdl, 2021-02-07:
Ich mach das alles bequem mit Sourcetree (https://www.sourcetreeapp.com). Kann ich uneingeschränkt empfehlen.
Gibt es für Win und Mac

TomMajor, 2021-02-07:
> Gerade wenn mehrere an einem Projekt arbeiten, sind eigene Branches (auf denen dann die Tests laufen und vor dem Merge ein Review gemacht wird) state of the art, oder?

Finde ich auch. Momentan hängen sehr viele HB Projekte am master branch. Ein develop branch mit Tests und idealerweise review nach 4-Augen Prinzip ist gängige Praxis.

pa-pa, 2021-02-07:
Deshalb hatte ich ja bisher immer die V(1|2|3|4)-Branches, damit ich den Master auch mal kaputt machen kann :-)
Aber die hat wahrscheinlich eh niemand wirklich genommen. 
Bei allen "schönen" Dev-Workflows (ich könnte da auch noch einiges reinwerfen) müssen wir aber sicherstellen, dass es noch zeitlich machbar ist. Ich wollte (wenn möglich) es erst mal einfach halten.

jp112sdl, 2021-02-07:
> Ich wollte (wenn möglich) es erst mal einfach halten.

Das wäre es doch mit Branches direkt hier, wie erst kürzlich mit dem "optimize"?

Das Problem mit den V-Branches ist, dass sie (nur) mit "älteren" Projekten funktionieren.

Änderungen im master erfolgen ja meist, um ein aktuelles Projekt lauffähig zu bekommen.

Und dann verweise ich auch in der Anleitung zum Projekt, zwingend den master-Branch zu verwenden.
Ich könnte auch auf einen bestimmten Commit verweisen, mit dem das jeweilige Projekt garantiert funktioniert. Das macht jedoch das Handling für Nachbauer schwieriger

Alternativ könnte ich bei meinen Projekten auf die Verwendung meines Forks verweisen.
Und den aktualisiere in unregelmäßigen Abständen oder nur bei Bedarf (nach vorheriger Travis-Prüfung).

pa-pa, 2021-02-07:
Extra-Fork fände ich eher schlecht. Habe Euch ja den Zugriff nicht gegeben, damit dann noch eine Kopie gepflegt werden muss.
Aber ich habe lieber ein aktives Projekt, wo dann sicherlich auch mal was kaputt geht, als wenn nicht passiert.
Jetzt wird die Lib ja auch über Arduino-Manager ausgeliefert. Vielleicht schafft das ja etwas "Luft", da ja nicht jede Änderung in der Lib zu einer neuen Version führt. Wir müssten dann halt sehen, das wichtige Anpassungen "zeitnah" als Version getaggt werden, damit der Library-Manager aktualisiert wird. Spricht natürlich auch wieder für Dev-Branches, wenn umfangreichere Änderungen vorgenommen werden.

TomMajor, 2021-02-07:
ich verweise auch wie Jerome teilweise auf den master Branch, die V4 ist ja relativ alt.
> Der HB-UNI-Sensor1 Sketch benötigt den master-Branch der AskSinPP Library, nicht den V4 Release-Branch wie dort angegeben. Grund: Verwendung der broadcastEvent() Methode.

IMHO ist es gängige und erwartete Praxis vieler github Projekte, das der master ein stabiler und verwendbarer branch ist.
Kleine überschaubare Änderungen kann man sicher direkt darin commiten.
Größere Änderungen wie ein neuer HW support sollten in einem dev branch fertig entwickelt werden, travis, gut getestet, review usw. und dann erst gemergt werden - just my 2ct.

jp112sdl, 2021-02-07:
Es ist ja nicht so, dass hier jeden zweiten Tag Änderungen einfließen.

Also letztendlich ging es mir hauptsächlich darum, Änderungen erstmal in einem extra Branch (egal ob im eigenen Fork oder direkt hier) durch den Travis laufen zu lassen, statt das direkt im Master machen zu lassen.

Ich war total verunsichert, weil ich die Änderung in der `Activity.h` genau aus dem Grund des vorherigen Travis-Run nicht direkt in den master gepusht habe. Der erste Versuch schlug auch tatsächlich fehl. Hab dann nachgebessert. Travis war ok.
Dann hab ich gemerged.
Ein paar Minuten später kam die Mail von Travis, dass es wieder Probleme gibt.
Und ich fing an, den Fehler bei mir und meinem Commit zu suchen.
Dass Trilu just in der kurzen Zeit was direkt in den master gepusht hatte, hab ich so schnell gar nicht realisiert.

@TomMajor Die V5 ist mittlerweile am nächsten dran am Master und hat auch `broadcastEvent`.

trilu2000, 2021-02-07:
Mir erscheint das mit dem 4 Augen Prinzip als nicht praktikabel, wer würde denn die Änderungen anschauen und final beurteilen?
Ich versuche mein Zeug ab sofort in einen eigenen Branch zu packen und als PR zu beschreiben - scheint ja als wären meine commits immer wieder der Grund des Problems...

jp112sdl, 2021-02-07:
Meine Commits funktionieren auch <strike>nicht immer</strike> nur selten auf Anhieb, wie z.B. hier bei mir heut früh:
[image]

Durch den PR (oder extra Branch hier) sieht man erstmal ob Travis noch alles bauen kann (derzeit werden **94 Sketche** geprüft!)

Wenn nicht - muss nachgebessert werden.
Ist der Haken grün, kann gemerged werden.

trilu2000, 2021-02-09:
falls jemand nach dem 4 Augen Prinzip kontrollieren möchte...
Simulated keypress implementation #232

jp112sdl, 2021-02-09:
Cool. Funktioniert das jetzt automatisch mit allen Aktor-Sketchen (Switch, Blind, Dimmer)?

trilu2000, 2021-02-09:
Momentan funktioniert das nur beim Dimmer - ich würde das aber in Switch und Blind einbauen, sobald ich wieder mal so ein Device brauche.

pa-pa, 2021-02-09:
Wieso funktioniert das nur im Dimmer ? Das ist doch alles in der Standard process() Methode.

jp112sdl, 2021-02-09:
Ich hab mal noch die README um `NDEBUG` und `HIDE_IGNORE_MSG` ergänzt

trilu2000, 2021-02-10:
@papa, Du hast recht, ich habe das mit dem power-up flag verwechselt, das ist nur im Dimmer implementiert.

pa-pa, 2021-04-25:
können wir jetzt zu machen

closed 2021-04-25 by pa-pa