pa-pa/AskSinPP #234: set PWM frequency in dimmer sketch

trilu2000, 2021-02-10:
I have started to implement the "Faktor PWM-Frequenz" within the dimmer classes.
Currently it works well at system boot - i can read the flag and set it within the pwms class.
But i have no idea how i can set it out of the configChanged event, either in DimmerDevice or DimmerChannel.

I didnt get access to the pwms class array, because it seems it is available only in the DimmerControl class.
Hopefully someone can push me in the right direction ...

Changes so far are in the dev-trilu branch

jp112sdl, 2021-02-10:
There are some dimmer sketches not working anymore.
```
$ buildSketch HM-LC-Dim1TPBU-FM
1357Picked up JAVA_TOOL_OPTIONS: 
1358Loading configuration...
1359Initializing packages...
1360Preparing boards...
1361Verifying...
1362In file included from /home/travis/build/pa-pa/AskSinPP/examples/HM-LC-Dim1TPBU-FM/HM-LC-Dim1TPBU-FM.ino:30:0:
1363/home/travis/Arduino/libraries/AskSinPP/Dimmer.h: In instantiation of 'bool as::DimmerControl<HalType, DimmerType, PWM>::init(HalType&, ...) [with HalType = as::AskSin<as::StatusLed<0>, as::NoBattery, as::Radio<as::AvrSPI<4, 5, 6, 7>, 10> >; DimmerType = as::DimmerDevice<as::AskSin<as::StatusLed<0>, as::NoBattery, as::Radio<as::AvrSPI<4, 5, 6, 7>, 10> >, as::DimmerChannel<as::AskSin<as::StatusLed<0>, as::NoBattery, as::Radio<as::AvrSPI<4, 5, 6, 7>, 10> >, 4>, 3, 3>; PWM = as::ZC_Control<>]':
1364/home/travis/build/pa-pa/AskSinPP/examples/HM-LC-Dim1TPBU-FM/HM-LC-Dim1TPBU-FM.ino:135:33:   required from here
1365/home/travis/Arduino/libraries/AskSinPP/Dimmer.h:809:15: error: 'class as::ZC_Control<>' has no member named 'setFreq'
1366       pwms[i].setFreq(m);
1367       ~~~~~~~~^~~~~~~
1368exit status 1
1369The command "buildSketch HM-LC-Dim1TPBU-FM" exited with 1.
```

and `$ buildSketch HB-LC-Dim1TPBU-FM_ISD2` also fails with the same error

trilu2000, 2021-02-10:
ups, class ZC_Control overseen
but it is only in dev-trilu, right?

jp112sdl, 2021-02-10:
Maybe... 😄 
Travis is down at the moment.
[image]

pa-pa, 2021-02-10:
You can simply set it always during updatePhysical again.

psi-4ward, 2021-02-10:
Ich bin mit asksinpp.de zu Github-Actions - das rennt soooo viel schneller. 
Ob es hier so Restriktionen gibt wie bei Travis weiß ich allerdings nicht.

trilu2000, 2021-02-11:
> You can simply set it always during updatePhysical again.

updatePhysical gets called every 10ms in dimmer and ~100 times for a change from 0% to 100%.
would mean i have to read and set List0().speedMultiplier() and List1().characteristic() 100 times in 0,5 sec transition time...

is there no way to trigger on a list change from outside? My understanding of the class structure and dependencies is getting better, but still not complete :-)

pa-pa, 2021-02-11:
> updatePhysical gets called every 10ms in dimmer and ~100 times for a change from 0% to 100%.
> would mean i have to read and set List0().speedMultiplier() and List1().characteristic() 100 times in 0,5 sec transition time...

I see no problem to read the list(s) here. Why you need to set values?

psi-4ward, 2021-02-11:
Bzgl CI gehts hier weiter: https://github.com/pa-pa/AskSinPP/discussions/236

trilu2000, 2021-02-12:
solved
pull request #239

closed 2021-02-12 by trilu2000