pa-pa/AskSinPP #259: Definition BME280 I2c address.

Plazmin, 2021-04-09:
I built my first homematic device by design [https://asksinpp.de/Projekte/psi/HM-WDS40-TH-I-BME280/#gehause](https://asksinpp.de/Projekte/psi/HM-WDS40-TH-I-BME280/#gehause).
I ran the sketch [https://github.com/jp112sdl/Beispiel_AskSinPP/blob/master/examples/HM-WDS40-TH-I-BME280/HM-WDS40-TH-I-BME280.ino](https://github.com/jp112sdl/Beispiel_AskSinPP/blob/master/examples/HM-WDS40-TH-I-BME280/HM-WDS40-TH-I-BME280.ino). Everything works fine, but only when the sensor has the 0x76 address set. If the sensor is set to address 0x77, the sketch does not read the temperature and humidity. Please let me know where and what I should define so that the sensor can work on the address 0x77. [](url)

jp112sdl, 2021-04-09:
The easiest way is to set the BME280 address to 0x76... You can use only 1 sensor with this sketch, so why u should use 0x77 ?

Other way ( not testet):
Change [line 99](https://github.com/jp112sdl/Beispiel_AskSinPP/blob/e4d695595e3a4d0451d66e58e548c09a6e7729d5/examples/HM-WDS40-TH-I-BME280/HM-WDS40-TH-I-BME280.ino#L99) to `Bme280<0x77>    bme280;`

and in the asksinpp-lib directory change [sensors/Bme280.h](https://github.com/pa-pa/AskSinPP/blob/master/sensors/Bme280.h) to
```
...
// https://github.com/finitespace/BME280
template <uint8_t ADDRESS=0x76>
class Bme280 : public Temperature, public Pressure, public Humidity {
  BME280I2C _bme;
public:
  Bme280 () {}
  void init () {
    Wire.begin();
    settings.bme280Addr=(ADDRESS == 0x76) ? BME280I2C::I2CAddr_0x76 : BME280I2C::I2CAddr_0x77;
...
```

Plazmin, 2021-04-09:
Thank you with a quick reply. Previously, I looked at the files you provided, but my knowledge is too little to find the correct solution myself. I think that in the evening I will have time to implement the changes you suggest.

I have a two questions: 
1. I would expect address definitions to be made directly in the arduino sketch. This is a parameter that the user should be able to set by himself. Why is there no such possibility and I need to make corrections deeper in libraries?
2. Where can I find information on how to start and what is the purpose of the second led from the basic device connection diagram [https://asksinpp.de/Grundlagen/01_hardware.html#funkmodul](https://asksinpp.de/Grundlagen/01_hardware.html#funkmodul)  In my device I have it connected according to the diagram, but unfortunately it does not signal anything.

Plazmin, 2021-04-13:
Any answer?

jp112sdl, 2021-04-13:
1. This is a community project. Neither the owner nor a collaborator has ever needed the option to set the address to 0x77.
So if you _"would expect address definitions to be made directly in the arduino sketch"_ you can create a pull request.

2. There is no information/documentation at all. As a hint: the second led is used in sketches with `DualStatusLed`

closed 2021-04-13 by Plazmin