pa-pa/AskSinPP #272: How to use this library with Platformio?

majuss, 2021-06-09:
Hi there, I'am trying to rebuild this example sketch: https://github.com/FUEL4EP/HomeAutomation/tree/master/AsksinPP_developments/sketches/HB-UNI-Sensor1-AQ-BME680_KF in Platformio. But I'am unable to compile, because PIO is unaware of most of the Arduino registers like:
```
.pio/libdeps/sparkfun_promicro16/AskSinPP/AlarmClock.h:221:19: error: 'TOIE2' was not declared in this scope
     TIMSK2  = (1<<TOIE2); //enable TOV2 interrupt
```
I already included <Arduino.h> but it's no use...
Here is my ini file:
```ini
[platformio]
src_dir = .

[env:sparkfun_promicro16]
platform = atmelavr
board = sparkfun_promicro16
framework = arduino
lib_deps = 
	AskSinPP
	erriez/ErriezCRC32@^1.0.1
	closedcube/ClosedCube BME680@^1.2.0
	rfetick/Kalman@^1.0.2
```

jp112sdl, 2021-06-09:
You have chosen a wrong AVR type:
`sparkfun_promicro16`
which uses an ATmega32U4, that is incompatible for this.

As described in the project README you have to use an ATMega1284P
https://github.com/FUEL4EP/HomeAutomation/tree/master/AsksinPP_developments/sketches/HB-UNI-Sensor1-AQ-BME680_KF#schaltung

jp112sdl, 2021-06-09:
> Is AskSinPP in general incompatible with the 32u4?

Yes. 

Supported hardware is listed in the AskSinPP README:
https://github.com/pa-pa/AskSinPP#asksin--

_(and don't mixup ATmega32 with ATmega32U4, they are completely different)_

TomMajor, 2021-06-09:
as an addition, this topic pops up from time to time in the HM forum, like here:
https://homematic-forum.de/forum/viewtopic.php?f=76&t=45083
(if you understand German)

jp112sdl, 2021-06-09:
here is an open issue, too: https://github.com/pa-pa/AskSinPP/issues/192

@TomMajor did "Bratmaxe" ever had success?

majuss, 2021-06-09:
So what is the best option when I want to build something like this device?
I only found that board: https://www.ebay.de/itm/254962772627?hash=item3b5cf72a93:g:KVoAAOSwdo5f3zKe
I first want to test a prototype and later on design a completely new highly integrated smd pcb with everything on one pcb including charge controller for Li-batteries and USB-C port (only for charging, if the 32u4 is not supported..).

jp112sdl, 2021-06-09:
> So what is the best option when I want to build something like this device?

I think, the best option is asking the project owner directly (i.e. by creating an issue there)
https://github.com/FUEL4EP/HomeAutomation/issues

Maybe you can use a breadboard and the 1284P (DIP package) for prototyping.

_I have built some prototypes of other homebrew devices with an ATmega644P / 1284P, based on this board:
https://github.com/JChristensen/mini1284_

majuss, 2021-06-09:
Thank you for the support I will bother the author! :)

closed 2021-06-09 by majuss