pa-pa/AskSinPP #275: Radio wakeup() dauert lange

jp112sdl, 2021-08-04:
Mir ist gerade bei einer Fernbedienung aufgefallen, dass nach dem Sketch-Sleep das Aufwecken des CC1101 (bei Verwendung einer CC1101-Abschaltung) 
https://github.com/pa-pa/AskSinPP/blob/91fadb267c3a71b75b500adf271e48bf4d408003/Radio.h#L1006
sehr lange dauert (fast 500ms).

Wenn ich in der `wakeup()` Methode statt
https://github.com/pa-pa/AskSinPP/blob/91fadb267c3a71b75b500adf271e48bf4d408003/Radio.h#L907
ein `init()` mache (und damit das komplette CC1101-init() durchlaufe) ist keine Verzögerung spürbar.

Ich finde leider absolut nicht heraus, warum `wakeup()` so lange dauert (was ja innerhalb seiner Methode auch nur ein `init()` aufruft?)... _Hab auch leider kein seriellen Debug am RWE BRC8 angelötet_

jp112sdl, 2021-08-04:
Wie es aussieht, reicht es schon, die Delays
https://github.com/pa-pa/AskSinPP/blob/91fadb267c3a71b75b500adf271e48bf4d408003/Radio.h#L454
und
https://github.com/pa-pa/AskSinPP/blob/91fadb267c3a71b75b500adf271e48bf4d408003/Radio.h#L504
auf 1ms runterzusetzen.

Würde aber auch bedeuten, dass irgendeine der Methoden mehr als 1x beim Aufwachen durchlaufen wird, weshalb sich diese große Verzögerung, wie im ursprünglichen Beitrag beschrieben, ergibt

pa-pa, 2021-08-04:
Hm - das ist komisch. Aber was mir gerade aufgefallen ist. Es wird im wakeup() und im init() gewartet. Das im wakeup() könnte odch eigentlich weg.

jp112sdl, 2021-08-05:
> Es wird im wakeup() und im init() gewartet.

Ja ich glaub das (und auch das doppelte `digitalWrite`) ist noch ein Relikt aus den ersten `wakeup()`-Versuchen.
Teste ich heut Abend mal.

jp112sdl, 2021-08-05:
> Das im wakeup() könnte odch eigentlich weg.

Ja, dem ist so. 👍

commit 2d32872b (2021-08-05): fixes #275
commit 711842f7 (2021-08-05): Merge pull request #276 from jp112sdl/master

fixes #275
closed 2021-08-05 by jp112sdl