pa-pa/AskSinPP #310: HM-ES-TX-WM: gas counter increases continuosly even if no input

soosp, 2023-03-05:
I tested HM-ES-TX-WM example configures as gas meter in the CCU. There was a reed switch attached to the Arduino pin 14 with proper pull up. When a magnet was approached to the reed switch the counter on the serial debug output displayed and increased properly. But the input in first reporting period wasn't displayed by CCU. From the second period the counter on CCU increases continuosly period by period enen if no input.

JurgenO, 2023-03-06:
Hi,
it worked for me. Are you sure you have connected A0 ? 
Please see my project description: https://github.com/JurgenO/GasMetering/wiki/%2303:-Build-the-hardware

soosp, 2023-03-06:
Yes, sure. It tesetd with reed switch and with pure contact only. It seems to be a software problem. If there is no input, the counter remains zero. But if a pulse arrives, the counter starts to incease.

soosp, 2023-03-06:
[image]

As you can see, only one pulse has arrived, the ratio is 0.01 m3/pulse, and after two periods the counter shows 0.002 m3. The counter was reseted before this test.

soosp, 2023-03-06:
[image]

Some minutes later...

jp112sdl, 2023-03-06:
Afaik, the 2-channel version in this repository only works correctly with FHEM.

For a ccu, I made a "1-channel only" version of the sketch:
https://github.com/jp112sdl/Beispiel_AskSinPP/blob/master/examples/HM-ES-TX-WM_CCU/HM-ES-TX-WM_CCU.ino

Give it a try...

soosp, 2023-03-06:
Thanks for the idea! It works with my CCU/RaspberryMatic. Only the Consumption value is strange for me. How is it calculated?

soosp, 2023-04-12:
Meanwhile I got that consumption unit is wrong in CCU. It should be m³/h.

closed 2023-04-12 by soosp