pa-pa/AskSinPP #316: Ignore repeated message

jp112sdl, 2024-01-16:
@pa-pa @trilu2000 

Mir war da gestern Abend in unserem Meeting noch was aufgefallen:
https://github.com/pa-pa/AskSinPP/blob/b83fd61f09f9a964eba8095ac5c5535606563b10/MultiChannelDevice.h#L247-L251

Das ist meiner Meinung nach falsch.
Hier wird nicht nur geprüft, ob die empfangene Meldung identisch zum vorherigen Absender+Counter ist, sondern auch ob `msg.isRepeated()` gesetzt ist.

Dieses Flag setzt afaik nur der Repeater (HM-Sys-sRP-Pl) und ist auch nur für diesen "interessant", um Endlosschleifen zu vermeiden.

https://github.com/pa-pa/AskSinPP/blob/b83fd61f09f9a964eba8095ac5c5535606563b10/Message.h#L63-L64

Also vom Flow her, wie ich es verstehe: 
Ein Gerät sendet eine Message mit `RPTEN`. Ein Repeater empfängt eine solche Message und erkennt, dass diese Nachricht überhaupt repeated werden darf. Anschließend wird von diesem das `RPTED` Flag angehängt, damit das Telegram nicht durch einen zweiten Repeater nochmals ausgesendet wird.

Gibt es gegenteilige Meinungen, `msg.isRepeated()` aus der Bedingungsprüfung zu entfernen?

_Ich glaub, das ist auch der Bug, der bei mir manchmal Aktoren mit verknüpftem Toggle-Fernbedienungstaster kurz ein und wieder aus schaltet. Es liegt wohl einfach nur daran, dass dann mehrfach gesendet._

trilu2000, 2024-01-16:
Hi Jerome,

Du hast recht, es ist ja völlig egal ob das repeated flag gesetzt ist oder nicht.

Eine wiederholte Nachricht bleibt eine wiederholte Nachricht und muss nicht noch einmal ausgeführt werden.

Einzig, es darf keinen Repeater Sketch geben 😊

Viele Grüße

Horst

 

https://github.com/pa-pa/AskSinPP/blob/b83fd61f09f9a964eba8095ac5c5535606563b10/MultiChannelDevice.h#L247-L251

https://github.com/pa-pa/AskSinPP/blob/b83fd61f09f9a964eba8095ac5c5535606563b10/Message.h#L63-L64

pa-pa, 2024-01-16:
👍

jp112sdl, 2024-01-16:
> Einzig, es darf keinen Repeater Sketch geben 😊

Doch, den gibt es :)
https://github.com/jp112sdl/Beispiel_AskSinPP/blob/master/examples/HM-Sys-sRP-Pl/HM-Sys-sRP-Pl.ino

Aber auch da ist es kein Problem, da Nachrichten, außer Config, nie an ihn adressiert sind. Der "hört" ja nur mit.

commit 05813667 (2024-01-16): fixes #316
commit c9e7eeb0 (2024-01-16): Merge pull request #317 from jp112sdl/master

fixes #316
closed 2024-01-16 by jp112sdl