pa-pa/AskSinPP #320: Possible endless loop in CC1101 timeout handling

andyboeh, 2024-02-02:
Hi,

I'm currently working on another project involving a CC1101 and I had a look at your radio routines. I think that you might have an endless loop in your code, effectively disabling the timeout:

Shouldn't the line

https://github.com/pa-pa/AskSinPP/blob/c7039d2e71231a736299707278b721ed0221044d/Radio-CC1101.h#L481

actually read
```
while ((spi.readReg(CC1101_MARCSTATE, CC1101_STATUS) != MARCSTATE_TX) && (--i != 0)) {
```

Otherwise, the code never times out, as the condition with the logical OR is still valid as long as the register doesn't have the correct value. NB: There are a few timeouts handled in the same way.

Please correct me if I'm on the wrong track here.

stan23, 2024-02-03:
Let's look at a few iterations. In the beginning `i` is 200 and state is not TX.

OR condition:
```
(someState != TX) || (199 != 0) = TRUE || TRUE = TRUE -> loop continues
(someState != TX) || (198 != 0) = TRUE || TRUE = TRUE -> loop continues
(someState != TX) || (197 != 0) = TRUE || TRUE = TRUE -> loop continues
(TX != TX) || (196 != 0) = FALSE || TRUE = TRUE -> loop continues unexpectedly
...
(TX != TX) || (1 != 0) = FALSE || TRUE = TRUE -> loop continues
(TX != TX) || (0 != 0) = FALSE || FALSE = FALSE -> loop breaks
```

or in case the TX state is never reached:
```
...
(someState != TX) || (0 != 0) = TRUE || FALSE = TRUE -> loop continues
(someState != TX) || (255 != 0) = TRUE || FALSE = TRUE -> loop continues endlessly
```

AND condition:
```
(someState != TX) &&(199 != 0) = TRUE && TRUE = TRUE -> loop continues
(someState != TX) && (198 != 0) = TRUE && TRUE = TRUE -> loop continues
(someState != TX) && (197 != 0) = TRUE && TRUE = TRUE -> loop continues
(TX != TX) && (196 != 0) = FALSE && TRUE = FALSE -> loop breaks!
```

or in case the TX state is never reached:
```
...
(someState != TX) && (1 != 0) = TRUE && TRUE = TRUE -> loop continues
(someState != TX) && (0 != 0) = TRUE && FALSE = FALSE -> loop breaks!
```

I agree with your analysis :-)

trilu2000, 2024-02-04:
many thanks for making me aware, i will fix it in the next days

trilu2000, 2024-02-07:
fixed!

closed 2024-02-07 by trilu2000