pa-pa/AskSinPP #323: AskSinPP working on Atmega4809?

stefanbbb, 2024-04-12:
Hi,
I would like to implement AskSinPP  on a platform which uses the Atmega4809. Currently this platform is not supported.
I compile to this chip with the MegaCoreX platform.
During compilation it turns out, that EnableInterrupt.h leads to errors. Due to the fact that AtMega4809 has interrupts on all pins, this library should not necessary for this platform.
Furthermore storage.h and LowPower.h are not supported

Is it planned to adapt the project to AtMega4809? This would help me a lot.

BR

pa-pa, 2024-04-14:
There is no plan to extend the library to AtMega4809. But I would be happy, if you do some work here and provide the PullRequest, so we can integrate this.
You can take a look on the STM32 or EFM32 related code. This also has no EnableInterrupt and LowPower.
https://github.com/pa-pa/AskSinPP/blob/7564545204d31cc11f841b1823e91355e9fb1620/AskSinPP.h#L47
https://github.com/pa-pa/AskSinPP/blob/7564545204d31cc11f841b1823e91355e9fb1620/Activity.h#L286

stefanbbb, 2024-04-14:
Hmm, I think that is too much for my skills. 
Another possibility:
The cc1101 gets its own Arduino Pro Mini, and this communicates via I2c or Serial with the main CPU (ATMEGA 4809)
Are there any existing status messages or protocolls, which I can use for this purpose?
Currently I see just the telegrams to CCU in the serial monitor.

jp112sdl, 2024-04-14:
> Another possibility:
The cc1101 gets its own Arduino Pro Mini, and this communicates via I2c or Serial with the main CPU (ATMEGA 4809)

What's your goal? Why is it not possible to realise your project with the pro mini only?

> Are there any existing status messages or protocolls, which I can use for this purpose?

No, there is nothing implemented to control asksinpp devices externally.

stefanbbb, 2024-04-14:
I have already a complex pcb also for other tasks, which is based on ATmega4809. I just like to add the homematic feature to this pcb. This is the reason why I cannot change to Pro Mini.

BTW: I'm just realizing that Homematic will not get produced any longer and prices for the remaining components are rising. This makes it difficult for me, because I wanted to mix self designed and purchased components in one project (RV control)
There is no possibility of direct coupling between Homematic IP and Homematic components, right?

stan23, 2024-04-14:
> There is no possibility of direct coupling between Homematic IP and Homematic components, right?

No, that’s not possible, neither for stock devices nor for selfmade devices.