pa-pa/AskSinPP #327: 2-button wall switch: Button as Variable without GPIO possible?

cablesky, 2025-02-16:
I want to replace an HM-PB-2-WM55 (2-button wall switch) with a keypad. 
The original button GPIO inputs should be deactivated. 
The goal is to virtually trigger the first button when the correct code is entered on the keypad. 
One key of the keypad has been connected to the Arduino's interrupt pin 3 to wake up the Arduino when someone wants to enter a code. 
The Arduino then checks the code. 
How can I trigger the switching process by changing a variable only, without using the GPIO pins? 

Does anyone have tips for implementation?

Thank you very much for your answer!

trilu2000, 2025-02-16:
Schau mal hier, ich habe sowas schon gebaut...
https://github.com/trilu2000/HB-CDL-6-Pinpad

cablesky, 2025-02-16:
@trilu2000 MEGA MEGA!! Wahnsinn! GENAU das hatte ich vor! VIELEN LIEBEN DANK! Hatte schon im I-Net danach gesucht, aber nix gefunden! PERFEKT!!! Ich freue mich!

trilu2000, 2025-02-16:
Freut mich wenn es dir hilft. Läuft bei mir seit 2 Jahren an der Haustür ohne Probleme. Selbst meine Frau nutzt das Ding und das will was heißen:-)
Melde dich wenn du Hilfe brauchst, oder auch Teile...

cablesky, 2025-02-16:
Danke für das Angebot, Dich fragen zu dürfen. Da wäre nämlich gleich die erste Frage: Ich habe so ein Keypad mit 12 Anschlüssen (also ohne Matrix - jede Taste hat einen eigenen Pin). Habe die Pinpad.h auch schon wie folgt angepasst. 

Aber eine Frage hätte ich noch. Welche Bibliotheken werden für Dein Code benutzt? 
Wollte mal kompilieren, da wurde die "key.h" nicht gefunden. Habe den Eintrag entfernt - aber jetzt werde ich mit Fehlermeldungen überschüttet, z.B. dass die Definition "HM_DEF_KEY" fehlt usw. 

Danke auf jeden Fall schon mal! 

-- Pinpad.h für 12-polige Keypads ohne Matrix ------

`#ifndef _PINPAD_H
#define _PINPAD_H

using namespace as;

static void* __gb_pinpad;

extern const int buttonPins[12];
extern const char buttonChars[12];

class Pinpad {
    static Pinpad& instance() {
        return *((Pinpad*)__gb_pinpad);
    }

    static void isr() {
        instance().handle_int();
    }

    void handle_int() {
        for (int i = 0; i < 12; i++) {
            if (digitalRead(buttonPins[i]) == LOW) {
                debounce.create_key(buttonChars[i]);
                debounce.start();
                keydown();
                return;
            }
        }
    }

    class Debounce : public Alarm {
        enum enum_state { IDLE = 0, WAIT, DEBOUNCE };
        volatile enum_state status;
        char key;
        Pinpad& pp;
    public:
        Debounce(Pinpad& p) : Alarm(0), pp(p), status(IDLE) {} 
        virtual ~Debounce() {}

        void start() {
            sysclock.cancel(*this);
            set(millis2ticks(50));
            status = WAIT;
            sysclock.add(*this);
        }

        enum_state state() {
            return status;
        }

        void create_key(char k) {
            key = k;
        }

        virtual void trigger(AlarmClock& clock) {
            bool keyPressed = false;
            for (int i = 0; i < 12; i++) {
                if (digitalRead(buttonPins[i]) == LOW) {
                    keyPressed = true;
                    break;
                }
            }
            if (!keyPressed) {
                if (status == DEBOUNCE) {
                    status = IDLE;
                    pp.haskey(key);
                    return;
                }
                if (status == WAIT) {
                    status = DEBOUNCE;
                }
            }
            set(millis2ticks(50));
            sysclock.add(*this);
        }
    } debounce;

public:
    Pinpad() : debounce(*this) {}
    ~Pinpad() {}

    void init() {
        __gb_pinpad = this;
        DPRINTLN(F("keypad init"));
        for (int i = 0; i < 12; i++) {
            pinMode(buttonPins[i], INPUT_PULLUP);
            enableInterrupt(buttonPins[i], Pinpad::isr, FALLING);
        }
    }

    virtual void keydown() {
        DPRINTLN(F("keydown"));
    }
    virtual void haskey(char k) {
        DPRINT(F("key: "));
        DPRINTLN(k);
    }
};

#endif
`

trilu2000, 2025-02-17:
Die key.h kannst du selbst bauen, da gehört nur eine Zeile mit dem AES Schlüssel rein.
`#define HM_DEF_KEY  0x01,0x02,0x03,0x04,0x05,0x06,0x07,0x08,0x09,0x10,0x11,0x12,0x13,0x14,0x15,0x16     `

Deine Pinpad Änderung sieht auf den ersten Blick ganz ok aus, aber da steckt der Teufel im Detail

cablesky, 2025-03-17:
Hallo Trilu, 
Ich habe jetzt das Keypad nachgebaut - und es läuft auch gut. Konnte es in Raspberrymatic einbinden. Läuft alles.
NUR:
Ich kann es nicht mit HomeAssistant synchronisieren. Es taucht einfach nicht auf. Andere Geräte von Dir (jp-hb) sind mit HomeAssistant schon synchronisiert worden.

Kannst du mir da evtl. weiterhelfen?